CTCaer/hekate #613: [Bug Fix]

quadsYT, 2021-05-12:
There's this thing called SD CARD Mis and it lets me use a usb connected to something and the switch to do sd card management. I tried this on chromebook, and when I ejected, it didn't stop. It instead kept going so i had to pull the cable..

Masamune3210, 2021-05-12:
What exactly are you talking about? The USB SD card Management in hekate? Or something else?

impeeza, 2021-05-12:
Could be SD Tools, UMS on Nix?  You cold update to Hekate 5.5.6 there is a update on UMS.  Then if the problem persists let us know and give more information about what are you doing, expected result and what happen

CTCaer, 2021-05-12:
Your chromebook then didn't do a medium eject. (That's an internal UMS command).
Basically hekate doesn't know how to identify an ejection from your Chrome OS.
In that case wait for your chromeboot to say that it's fine to remove it and then pull the cable or press VOL- and VOL+

(heuristic support for such OS will not be added because I don't have a device to test possible heuristics.)

closed 2021-05-12 by CTCaer