CTCaer/hekate #614: Exception Occurs when Booting Hekate from TegraRCM-GUI

TheModdersDen, 2021-05-12:
I'll make this brief, but as detailed as possible:

Updated to FW 12.0.2 on accident today. Due to that, I updated to the latest Atmosphere (1.19.3), and tried to update to the latest Hekate to solve booting problems with the new FW version pushed out by our good friend, Nintendo.

Turns out, I get this error when trying to launch from my laptop with TegraRCM-GUI with the latest Hekate payload:
"An exception occured (LR 8105A4E8):
UNDEF

Press any key..."

Upon pressing any of the 3 keys (power and both volume buttons), I am brought back to this same page. I CAN boot my switch with the fusee-primary.bin payload, but cannot use Hekate's payload for some reason.

Things I have tried:
- Turning my switch off and on again
- Making sure that my SD card had the latest Hekate files on it (overwriting the new ones over the old ones)
- Using a different payload (fusee-primary.bin) to see if it's the Hekate payload that's the issue (as previously mentioned). It would seem as if that's the case.

Some additional notes:
1. My SD card is formatted to exFAT
2. I overwrote the files in the bootloader folder, but noticed that it had some leftover files that weren't overwritten.
3. I would prefer not to wipe my SD card as I have a bunch of personal, LEGAL game backups of games that I own on there that took a lot of time and effort to put on the SD card. I'd prefer not to go through that again.

Any ideas, thoughts, or suggestions would be useful and are welcome.

Thanks so much,
TheModdersDen

impeeza, 2021-05-12:
You don't need to clean ALL the card but I strongly recommend to backup and clear the atmosphere & bootloader folders and then recreate from the latest version of atmosphère and Hekate.  Create your hekate_ipl.ini with the needed configuration and try.  If the system starts. Begin to install sysmodules you need, and other homebrew what copies things to atmosphere's folder and discard if is some unsupported homebrew the offender

CTCaer, 2021-05-12:
@TheModdersDen 
Copy the contents of hekate_ipl.ini and whatever else you can't find to replace again.
Delete the corrupted bootloader folder and paste anew.
create a hekate_ipl.ini and paste the old contents, assuming they are not corrupted.
Safely eject and inject.

TheModdersDen, 2021-05-12:
@CTCaer Thanks. I am not able to give that a shot at the moment, but will let you know as soon as I can.

TheModdersDen, 2021-05-13:
@CTCaer Thanks, what you said worked! :)

closed 2021-05-13 by TheModdersDen