CTCaer/hekate #615: emunand loading problem

theEofE, 2021-05-15:
i just created my emunand via SD Partition, and every time I boot on to the "CFW (emunand)" I get 

"Unknown pkg1 version.
 HOS version not supported!
 Or emuMMC corrupt!
 Trying to backup bootloader...
 found pkg1 (//).

 Unknown pkg1 version
 HOS version not supported
 Or emuMMC corrupt
 
  Failed to launch HOS!"

is it because I have a fake or defective SD card? here is the info of my SD card: 
[image]

urherenow, 2021-05-16:
Yup. Your card is fake, and/or your card slot is broken.

theEofE, 2021-05-16:
> Your card is fake, and/or your card slot is broken.

oh ok

CTCaer, 2021-05-16:
Card is 100% fake.

closed 2021-05-16 by CTCaer