CTCaer/hekate #616: Failed to match warm boot with fuses - Sleep Mode won't work

Moriarty92x, 2021-05-15:
I'm having some trouble after updating to the latest Atmosphere. I have a Mariko Switch.

I updated Atmosphere, Hekate and sigpatches to latest versions. Updated emunNAND to 12.0.1 with daybreak. Reset, everything worked fine. Later I updated sysNAND to 12.0.2 and now I keep getting this error when I try to boot emuNAND: "Failed to match warm boot with fuses - Sleep Mode won't work". I'd really appreciate some help.

CTCaer, 2021-05-16:
Go to tools -> 2nd tab -> dump pkg1/2 and tell me the build date it will show.

Moriarty92x, 2021-05-17:
Found pgkg1 ('20210422145837')

Is this it?

CTCaer, 2021-05-17:
If that's the build date, then you should not get that message.
What Burnt Fuses (ODM 7/6) and ODM Fields (4, 6, 7) say in Console info -> HW & fuses?

me2151, 2021-06-13:
I have the same issue, similar setup, mariko, 12.0.3 emunand, same date string for pkg1 ('20210422145837'), Burnt fuses (ODM 7/6) : 15-0 (HOS: 12.0.2+), ODM Fields (4,6,7): 00010A40 00000000 00007FFF

CTCaer, 2021-06-13:
Boot once stock or sys cfw and it will be fixed. It will dump the correct warmboot fw.
(The 20210422145837 value you got from Dump pkg1/2 is from sys. Not emu.)

Also even then, your emu pkg1 is something else (you can see it if you set Logs ON in hekate before booting emu). 
This will create other issues because pkg1 build date is used for setting up stuff on boot.
**You need to use an unbrick guide or reupdate to 12.0.3 in order to update/fix your pkg1 in BOOT0.**
If you go the reupdate path and your emu's pkg1 is still an older version then there's something wrong with the update tool you used and you should report it to its dev.

closed 2021-06-13 by CTCaer

CTCaer, 2022-08-05:
@NachoPSanchez The HOS version you are trying to run is 10.x.x.
It doesn't have the required warmboot firmware you need.

You've run 12.x or 13.x in the past so you had the firmware at some point but you deleted it. (you must always safekeep the warmboot_mariko folder)

You have 2 options;
1. skip the warning and boot like that and update HOS
2. create an emummc and update it to [choose between 12.0.2 to 13.20] and boot it once to get the firmware.

NachoPSanchez, 2022-08-06:
Okey thank you so much! 
1. And what happens if i boot like that? I tryied once and some nintendo error appears, so i refused trying another time. 
2. I have emummc but how can i update without booting?

About safekeep the warmboot_mariko folder, i have all the files from the old sd, but i hadnt this folder in the past i think because i was using sx os

[image]

urherenow, 2022-08-07:
ChoiDujour has not been the preferred method of installing firmware upgrades for quite a while. Sept no longer exists. It's highly unlikely that you EVER used sxos on a Mariko. You are creating more problems for yourself, that have nothing to do with Hekate or Atmosphere bugs. This isn't a support forum.

 
 

Okey thank you so much!

CTCaer, 2022-08-07:
@NachoPSanchez
1. You have other issues if you saw a nintendo error. The only thing that breaks is deep sleep. Which means if you enter it, it will stay there.
2. Already told you how (unbrick guide) if you can't boot (assuming you tried without sysmodules and stuff).

But anyway that's not a hekate issue.

NachoPSanchez, 2022-08-12:
Thank you so much for the answer, i tryied a lot of thing for not do the first option because im trying not get a ban. 
Im trying to enter to emmumc but this error appears on the screen:
"A fatal error ocurred when running Fusée.
Failed to locate wamboot firmware"

I checked the sd and it have a warmboot firmware folder with a file. We talked about that folder and you told me to not delete this folder, but i never had this folder on my sd because its the first time for me trying to boot Atmosphere because i was using sx os.

Arthurlelynx, 2023-06-17:
hello i have the same problem my switch says Failed to match warm boot with fuses - Sleep Mode won't work my version of the PKG 1/2 is 20230111100014 thank you in advance for your answer

sorry for my bad english it's not my mother tongue

son1123, 2023-08-06:
hello i have the same problem my switch says Failed to match warm boot with fuses - Sleep Mode won't work.

Help me please

CaptainReefr, 2023-09-17:
Whenever any issues seemingly arise from nowhere, update Hekate, Nyx, Atmosphere and sig patches...THEN update the official firmware (assuming the latest hekate and atmosphere support it...if not...then wait for those updates)

From the Hekate menu, boot into OFFICIAL FIRMWARE.

That guy said booting official will fix it, but for me, it didn't work unless I booted into official from the Hekate menu...this error went away, and furthermore my Switch's chip stopped taking forever to activate.  I would watch the purple pulsing light for ages and then it would either flash orange/yellow or sometimes turn red and boot into official mode, skipping hekate altogether...after retrying several times, Hekate would sometimes boot....very scary, but after updating and booting into official from the Hekate menu when it finally booted, Hekate is starting up quickly as normal everytime.