CTCaer/hekate #617: Backup BOOT0/1 from eMMC and restore to (migrated SX OS) emuMMC doesnt work

felipejfc, 2021-05-15:
Context:
- Patched Erista running 10.2 FW and with SX Core

I'm using SX Core to boot and I wanted to migrate from sx os hidden partition emunand to atmosphere, here's what I did:

- Booted switch into SX Menu and did Cleanup
- Booted hekate from SX Menu Payload

I wanted then to backup my eMMC BOOT0 and 1 partitions and to restore to the emunand so that I ensure they're not corrupted. Backing up worked fine.

I then got BOOT0 and BOOT1 backup files and put them inside restore dir, then I went to hekate and tried to restore my emuMMC BOOT0/1, but I get this error:

[image]

This is an sx os hidden partition emunand that I migrated using hekate, it let me migrate and shows emunand as active, so I don't get why it refuse to restore the partitions.

Any hint?

I ended up restoring with NXNandManager, however I couldn't get the emunand to boot with atmosphere yet, it stays in a black screen after showing the blue atmosphere splash screen, then I'm investigating this issue since it may reveal that maybe hekate is failing to correctly read the emunand somehow?

CTCaer, 2021-05-16:
Only partition based emuMMC can be backed up and restored in hekate.

closed 2021-05-16 by CTCaer

luanzanetich, 2021-11-24:
hey! did u discovered what was happening? here i got the same black screen issue after restoring de boot0/boot1.

impeeza, 2021-11-25:
SXOS es know for damage the sysnand files including the BOOT0 & BOOT1,  search the net, about how to repair them there is several guides, this is not space for support,  on GBATemp there is a thread exclusively to that problem, you can also ask on discord.