CTCaer/hekate #618: Compiling error

benevido, 2021-05-22:
Hello I don't succeed to create hekate.bin 
What is the problem in my windows 10 configuration ?
I have devkitpro 3.0.3 but 
I have these mistakes after the make command in my powershell :

./bootloader/frontend/fe_emmc_tools.c:920:9: warning: #pragma GCC target is not supported for this machine [-Wpragmas]
#pragma GCC pop_options
^
cc1.exe: warning: unrecognized command line option '-Wno-stringop-overflow'
cc1.exe: warning: unrecognized command line option '-Wno-stringop-overread'
Building build/hekate/fe_info.o
./bootloader/frontend/fe_info.c:665:9: warning: #pragma GCC target is not supported for this machine [-Wpragmas]
#pragma GCC pop_options
^
cc1.exe: warning: unrecognized command line option '-Wno-stringop-overflow'
cc1.exe: warning: unrecognized command line option '-Wno-stringop-overread'
Building build/hekate/fe_tools.o
./bootloader/frontend/fe_tools.c:609:9: warning: #pragma GCC target is not supported for this machine [-Wpragmas]
#pragma GCC pop_options
^
cc1.exe: warning: unrecognized command line option '-Wno-stringop-overflow'
cc1.exe: warning: unrecognized command line option '-Wno-stringop-overread'
Building build/hekate/ccplex.o
Building build/hekate/clock.o
Building build/hekate/di.o
Building build/hekate/gpio.o
Building build/hekate/irq.o
./bdk/soc/irq.c:238:1: warning: target attribute is not supported on this machine [-Wattributes]
{
^
./bdk/soc/irq.c:261:1: warning: target attribute is not supported on this machine [-Wattributes]
{
^
./bdk/soc/irq.c: In function 'fiq_handler':
./bdk/soc/irq.c:273:1: error: interrupt Service Routines cannot be coded in Thumb mode
}
^
./bdk/soc/irq.c: At top level:
cc1.exe: warning: unrecognized command line option '-Wno-stringop-overflow'
cc1.exe: warning: unrecognized command line option '-Wno-stringop-overread'
make: *** [Makefile:143: build/hekate/irq.o] Error 1
PS F:\Games\SWitch\hack perso\hekate-5.5.6>

urherenow, 2021-05-22:
is your environment set correctly?

did you install dkp-pacman -S switch-dev?

Did you ensure everything is up to date? dkp-pacman -Syu

This isn't a help forum. It's for reporting bugs. Please use Discord or gbatemp.

CTCaer, 2021-05-22:
Your environment is incorrect.
Dunno what compiler you used but it does not support GCC pragmas.
More importantly, your actual error is because you disabled arm instructions completely or something.

Follow the instructions above to fix it or instruct a dkp site.

closed 2021-05-22 by CTCaer

benevido, 2021-05-22:
> Your environment is incorrect.
> Dunno what compiler you used but it does not support GCC pragmas.
> More importantly, your actual error is because you disabled arm instructions completely or something.
> 
> Follow the instructions above to fix it or instruct a dkp site.

Thanks I do post in gbatemp and discord but no answers. So I try here, thanks for your time, I ll use your informations to solve my issue.