CTCaer/hekate #619: Error Code LR 8105a520

Nelson20241, 2021-05-28:
I'll make this brief, but as detailed as possible:

I was trying to update to FW 12.0.2 from 10.0.2. I was trying to use [AIO updater](https://github.com/HamletDuFromage/aio-switch-updater/releases), I updated to the latest Atmosphere (1.19.3) and tried to update to the latest Hekate to solve booting problems with some games that I had installed.

When the program finishes downloading both Hetake and Atmosphere and reboots I got this error:

"An exception occurred (LR 8105a520):
UNDEF

Press any key..."

Upon pressing any of the 3 keys (power and both volume buttons), I am brought back to this same page. I CAN boot my switch normally but every time I inject the Hetake payload from TegraRCM on my PC I get the same message.

Things I have tried:

Turning my switch off and on again
Making sure that my SD card had the latest Hekate files on it (overwriting the new ones over the old ones)
Using a different payload (fusee-primary.bin) to see if it's the Hekate payload that's the issue.
I tried googling the issue but it seems it's not very common since I could not find any threads on Reddit, GBAtemp, and here.

Some additional notes:

My SD card is formatted to exFAT
I overwrote the files in the bootloader folder but noticed that it had some leftover files that weren't overwritten.
I would prefer not to wipe my SD card as I have a bunch of personal, but if it has to be done then I have no choice.
I am currently using AutoRCM.
Any ideas, thoughts, or suggestions would be useful and are welcome.

Thanks so much,
Nelson20241

impeeza, 2021-05-28:
Is recomended always to update at hand, is easy and you have the control of the keeping your configurations.

download the latest files and update on your SD by hand.

CTCaer, 2021-05-29:
Your sd is corrupted.
Follow https://github.com/CTCaer/hekate/issues/614 to fix the directory fat record.
Or simply format it (because guaranteed you have more corruption).

closed 2021-05-29 by CTCaer