CTCaer/hekate #620: Freeze on selecting SD Card USB Mass storage

corang, 2021-06-06:
Hekate freezes upon selecting the sd card UMS tool. I've tried from the partition manager area as well as from the tools area. I have also tried multiple cables and ports on my pc. Hekate is loaded via the guide at switch.homebrew.guide with the fusee-gelee method. SD card is a 256gb Samsung.

corang, 2021-06-06:
I do get a popup window saying
```
USB Mass Storage
Device: SD Card
Status: Started USB
```

The only thing I'm able to do is hard shutdown the switch

CTCaer, 2021-06-06:
Edit `bootloader/nyx.ini` and change `bpmpclock=1` to `bpmpclock=2`. (If it was already 2, tell me).
Reply with results.

corang, 2021-06-06:
> 
> 
> Edit `bootloader/nyx.ini` and change `bpmpclock=1` to `bpmpclock=2`. (If it was already 2, tell me).
> Reply with results.

Same result, did an entire reboot to double check.

CTCaer, 2021-06-06:
Are you sure it freezes then? What happens in that Started USB screen if you press VOL- and VOL+ or if you remove the cable?

corang, 2021-06-06:
Pretty sure, dumping a game right now then I'll check, the way the screen is dimmed it looks like it never even finishes the animation of bringing up the popup window.

corang, 2021-06-06:
Yeah definitely a freeze, no input changes anything except for holding power button. Unplugging the cable and starting it without the cable doesn't do anything different either. I can take a video if that would help?

corang, 2021-06-06:
The UMS tool in TegraRcmGUI doesn't work either

CTCaer, 2021-06-13:
If the ums tool in tegrarcmgui (which uses memloader and uboot) doesn't work then you probably have a hardware issue and there's nothing I can do.

closed 2021-06-13 by CTCaer

vos6434, 2021-06-13:
It worked on an older version of Hekate but then I updated both cfw and hekate and that's when it also has stopped working for me.

olivatooo, 2022-12-26:
Same here... idk why it freezes

cnelfalk, 2025-12-31:
Years later, and I can confirm this also happens to me.

swankleey, 2026-01-06:
Literally just made an account because I had this same issue (or a very similar issue) and this is how I fixed it. For me, the issue occurred when I went to USB Tools, went to USB Mass Storage, and clicked SD Card. A pop-up would appear, and then the screen would suddenly dim and completely freeze.

The pop-up displayed the following:
```
USB Mass Storage
Device: SD Card
Status: Started UMS
Note: To end it, safely eject from inside the OS.
DO NOT remove the cable!
```

I figured out the reason why it freezes is actually intentional (at least, I think it is). It doesn't want you to close it out unless you go to your file manager, go to your microSD drive (if it is not popping up, make sure you have nothing plugged into your USB-C port on your PC. That was my main issue. It was my MicroSD card reader thingy, but I had taken the MicroSD card out already, so it wasn't actually doing anything. I removed it, and suddenly, the MicroSD drive popped up.), and click the eject button at the top so you eject your MicroSD safely. Once you eject it, your switch should unfreeze and you can click the close button. I don't know if this solution will help everyone, but I couldn't believe after all these years, no one had a solution.

Images provided are of the eject button and the MicroSD card reader thingy that was plugged into my USB-C port that made it so I couldn't see my MicroSD card drive (well, it's not the exact same thing, but it's close enough.)

[image]

[image]

Please let me know if this helps!