CTCaer/hekate #622: Emummc restriction failed

bjgamer2019, 2021-06-14:
I normally use the option `emummcforce=1` to avoid boot on SysNAND and always force the emuMMC usage. If I don't have an emuMMC, boot fails (as expected and desired).

The last release (5.5.7) apparently does not respect the restriction and boot on SysNAND with CFW even if the option is there. Part of my `hekate_ipl.ini`

```
{-- Custom Firmwares --}
[AMS 0.19.4 (EmuNAND)]
emummcforce=1
payload=bootloader/payloads/fusee-primary.bin
{ }
```

If need any extra information, let me know. Thanks for your good work.

Masamune3210, 2021-06-14:
Someone correct me if I'm wrong but I'm pretty sure variables like that don't apply if you use a payload to boot instead of booting directly with hekate

CTCaer, 2021-06-14:
> Someone correct me if I'm wrong but I'm pretty sure variables like that don't apply if you use a payload to boot instead of booting directly with hekate

correct

closed 2021-06-14 by CTCaer

bjgamer2019, 2021-06-14:
> Someone correct me if I'm wrong but I'm pretty sure variables like that don't apply if you use a payload to boot instead of booting directly with hekate

I inject `hekate` and chainload with `primary-fusee`. On old versions that worked, but if that's not the right way, I will verify how to force it back again.