CTCaer/hekate #624: Help, i'm in danger

K0smic, 2021-06-15:
I wanted to restore my emummc but I did NOT activate SD mode (it was off) so I don't know what I restored.
The problem is that my switch is an ipatched so I'm afraid to reboot or shut down the switch.

K0smic, 2021-06-15:
I think I did a restore of the emumcc inside the mmc and I don't know what that entails.
I have a backup of the sysnand (4.1) but not the BOOTs (I have two boot files inside the emummc backup)

CTCaer, 2021-06-15:
Read and reread the following carefully and do not skip sentences.

- Do not reboot otherwise it's done for. Especially if BOOT0/1 do not match GPP (which means a soft brick).
So quickly restore the 4.1 GPP backup and find the boot0/1 files to restore them also.
If you don't have them then you will need to recreate them from scratch via an unbrick guide.
You will also need to recreate the boot0 keyblobs because low versions depend on them. And these are console specific.

- If you don't have your console keys to recreate the keyblobs, then it's done for and full emummc backup restore on top of emmc is the only way (make sure that emummc boot0 does not have autorcm and matches emummc GPP).

Make sure it doesn't get out of battery juice while you are at it.

Allowed options for you in hekate is Reload nyx only. Any reboot button or power off are to be avoided until your issue is fixed.
Whatever you do, triple check that boot0/1 (bct, pkg1, keyblobs) are fine and GPP matches them.

These are the only advices I can give, because here's not a support forum.

closed 2021-06-15 by CTCaer