CTCaer/hekate #627: Hekate freeze when i press usbtools=> sd card

zaikozako, 2021-06-26:
Hello whenever i wanna connect my switch to pc (just setting up)
I enter usbtools=> sd card
And it says started usb but then my switch is freeze/hardstuck/softlock
And i can only reboot
How can i fix this?
(Btw dbi-awoo work with usb but not hekate)

CTCaer, 2021-06-26:
Edit /bootloader/nyx.ini and change `bpmpclock` to `2`.
Tell me what happens.

zaikozako, 2021-06-26:
It worked thanks a lot

CTCaer, 2021-06-26:
Great!

closed 2021-06-26 by CTCaer

Temporalin, 2023-10-14:
Hi! I just had this problem after updating Hekate, and I don't really understand why I needed to change that obscure option. Could you point out what could be the cause? Just out of curiosity

CTCaer, 2023-10-15:
@Temporalin Very low binning at that part of the SoC.
You probably already had that setting automatically before and you updated by deleting the bootloader folder.
Which you should never do for hekate except if the folder is corrupted.