CTCaer/hekate #630: Cannot create a SD emuMMC backup

unknowg, 2021-07-03:
Whenever I try to create emuMMC backup I'm getting "Failed to find a partition... Failed!" error on BOOT0/BOOT1 and "Failed to find a partition... #FFDDOO Failed!" on RAW GPP. My firmware version is 12.0.1 and Hekate version is 5.5.7. Also my SD card have 86GB of free memory

closed 2021-07-03 by unknowg

alexnovelli, 2021-11-11:
@unknowg I'm having the same problem here. How can you resolve that? Tks!

urherenow, 2021-11-11:
Have you tried deleting your Atmosphere folder and dropping in a fresh one from the latest release .zip? IIRC, there were some things in older versions that were removed and/or moved, but might still be getting loaded if they exist in the content sub folder...

Masamune3210, 2021-11-11:
nothing in the atmosphere folder is going to be affecting anything in hekate unless you are booting atmosphere

urherenow, 2021-11-11:
Lol... wrong. Whatever replaced SEPT is NEEDED to access emmc.

impeeza, 2021-11-11:
Sept folder is no more needed  nor by Atmosphère nor hekate. Delete that folder, nachos and delete atmosphere folder and copy new u one from latest release. This space is not for support for supper go to discord or gbatemp. AND SEARCH THERE IS PLENTY OF ALREADY THREAD ABOUT THIS PROBLEM