CTCaer/hekate #632: Hekate freezing (typically on the logo screen)

karlland11, 2021-07-08:
Hekate would freeze up completely after pushing the payload.  Most of the time it would freeze at the logo screen, rarely would it freeze up after 5 seconds into the home menu, every time it would freeze.  I ended up ruling out that the problem is related to my micro sd card because the exact same micro sd card worked perfectly fine in a different hacked switch.  

I was able to find that v5.0.0 works fine with my switch.  I was also able to find that in the newest version of hekate if I removed nyx.bin I was able to boot hekate (though the features were very limited i.e no emummc creation).  

I am able to run nintendo's firmware fine with no problems/delays on this switch.

CTCaer, 2021-07-08:
I wonder why you managed to get pass over my 5s check without a hang on nyx boot.
5.0.0 in that case might work because it doesn't have joycon driver. And uart adds more stress on SCLK.

Do this on latest nyx and tell me what happens:
Edit /bootloader/nyx.ini and change `bpmpclock=` to `2`.
Reinject hekate from start and tell me what happens.

Also how relatively new is your Switch? Because right now I have around 6-7 reports with that and all of them are somewhat new for an Erista.

karlland11, 2021-07-08:
"Do this on latest nyx and tell me what happens:
Edit /bootloader/nyx.ini and change bpmpclock= to 2.
Reinject hekate from start and tell me what happens."

nyn.ini was already set to bpmpclock = 2.  (Maybe I misunderstood something)
I reinjected hekate from start and ran into the freeze problem.

I did some testing:
All versions above v5.1.4 would freeze immediately.
v5.1.4 froze after 10-30 seconds
v5.1.3 froze after 10-30 seconds.  Would freeze sooner if I messed with the create emuMMC button.
*from here on the tests will be done by going to create emummc then 'cancel' and repeat until hekate freezes (this is when I stop the timer).
v5.1.2 froze after 15 seconds.
v5.1.1 froze after 4 seconds.
v5.1.0 froze after 5 seconds.
v5.0.2 did NOT freeze.

I don't know how new my switch is.  Check the RetroNX hekate support channel for more information on what I said regarding this. 

Both nyx.bin files you sent over did not work with the newest bootloader/hekate.  Hekate would freeze on the logo screen.

You recommended "but can't give a build now but you can still try that by deleting or renaming libsys_minerva.bso"
This gave me a "missing or old minerva lib!" error.  When I press any key it brings me to a frozen hekate logo.

Again, hekate boots fine when nyx.bin is removed from the newest version v5.5.8.  What is strange is that the things you disabled and had me test out were no solution.

CTCaer, 2022-05-17:
Closing this. If you want to reopen it, in case you still think it's still not a HW issue feel free, but test the latest hekate/Nyx beforehand.

closed 2022-05-17 by CTCaer