CTCaer/hekate #633: Hekate 5.5.8 + Atmosphere 0.19.5 -> Black screen when injecting payload

KikoEsUnDios, 2021-07-08:
Hi everyone

I've never got this problem but I tried to play today and this happened to me.

I'm using the latest version of Hekate and Atmosphere with the swith in 12.1.0, and after injecting the fusee-primary through Hekate, Atmosphere logo pops up, Nintendo logo pops up, and then... black screen.

I'm using a 128 Gb SD card and I'm doing a partition with Hekate (89 and 29), and putting the emuMMC in that partition (FAT32), but it keeps failing. When I try to see what's in this partition on my PC, it seems like the 29Gb partition is damaged and i have to format it.

Any ideas on how can I fix this? Thanks :)

CTCaer, 2021-07-08:
You can't mount that partition in windows and do not format.
Other than that you are not using hekate to boot, so that issue report is in the wrong repo.

If you want you can try to boot via hekate. If it still doesn't work then reopen the issue.

closed 2021-07-08 by CTCaer

KikoEsUnDios, 2021-07-08:
Sorry to reopen.

I think I'm using Hekate to boot (Launch -> Atmosphere). And I don't understand what you meant with "You cant mount that partition on Windows". I'm doing the partition through Hekate, putting emuMMC to 29(full) and it seems to work. The problem is when I try to do the emuMMC, thats when i put my sd in my pc and it seems like its not ok that partition

Masamune3210, 2021-07-08:
If you are booting through the payload= or through the payloads screen you are not using hekate to boot

If you are setting your entire SD card to be a emunand then it is normal for the sd card to no longer be able to be read from pc