CTCaer/hekate #634: Hekate 5.5.8 + Atmosphere 0.19.5 -> Black screen when injecting payload

KikoEsUnDios, 2021-07-08:
Sorry to reopen.

I think I'm using Hekate to boot (Launch -> Atmosphere). And I don't understand what you meant with "You cant mount that partition on Windows". I'm doing the partition through Hekate, putting emuMMC to 29(full) and it seems to work. The problem is when I try to do the emuMMC, thats when i put my sd in my pc and it seems like its not ok that partition.

impeeza, 2021-07-08:
when you create a emunand partition, a RAW partition is created on your SD card, Windows thinks what that RAW partition is damaged and alerts you to "repair" but if you repair that partition all emunand data is lost, so please DO NOT ALLOW WINDOWS TO "REPAIR" your SD, you can use several tools on windows to backup and restore that partition I personally love  [EmuTool](https://gbatemp.net/threads/emutool-move-partition-emu-on-another-sd-switch-emu-type-on-sxos-and-more.550756) when you have Hekate and NYX working I strongly recommend not to remove the SD card but use the integrates UMS connection on NYX to connecti your SD Card to computer, you will avoid lot of trouble with the switch card reader.

KikoEsUnDios, 2021-07-09:
OK. I will not repair it.

so.... I'm supposed to have..?
- A 89Gb partition with all the directories and files there (atmosphere, hekate, .nro, etc)?
- A 29Gb partition RAW for emunand

If yes, what could be the problem to the black screen issue?. I think I'm doing it ok. 
1. Create partition through Hekate
2. Create emuMMC on the partition 1 (29Gb)
3. Launch Atmosphere through Hekate (Launch menu)
4. wait... -> Black screen

Thanks btw :)

CTCaer, 2021-07-09:
Show me the contents of hekate_ipl.ini
Use the following syntax to enclose it.
\`\`\`
copy pasted text
\`\`\`

KikoEsUnDios, 2021-07-09:
Before or after launching the Atmosphere?

What I meant to say it's if that the "hekate_ipl.ini" it's modified in some point of the configuration, or it's always the same file and it's not modified in any moment.

impeeza, 2021-07-09:
Hi, Hekate_ipl.ini is a file **you** must to personalize to fit your need, that's why the "prepacked" packages not always are good idea, please let us to know your ini file to diagnose if a bug or a miss-configuration.

KikoEsUnDios, 2021-07-09:
```
[config]
autoboot=0
autoboot_list=0
bootwait=0
backlight=220
autohosoff=0
autonogc=1
updater2p=1

{AtlasNX/Kosmos v15.2}

{}

{Discord: discord.teamatlasnx.com}

{Github: git.teamatlasnx.com}

{Patreon: patreon.teamatlasnx.com}

{Pegascape DNS: pegascape.sdsetup.com}

{}

{--- Custom Firmware ---}

[Atmosphere (EMUMMC)]
payload=bootloader/payloads/fusee-primary.bin
icon=bootloader/res/atmoslogo.bmp
logopath=bootloader/bootlogo.bmp

{}

{--- Stock ---}

[Stock]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1
icon=bootloader/res/icon_switch.bmp

{}

```

impeeza, 2021-07-09:
Are you using additional modules like Mission Control themes or similar?  some of them has been updated to avoid Black Screens on boot on atmosphere 0.19.5

KikoEsUnDios, 2021-07-09:
I don't know what "Mission Control module" is, but if it solves the problem, let's try it :)

What is it? How do I use it?

KikoEsUnDios, 2021-07-09:
Nothing. I download the last version of Mission Control (0.5.0), put it the root of the SD card, and still the black screen. :(

CTCaer, 2021-07-09:
You are not using hekate to boot.
As I said:
>Other than that you are not using hekate to boot, so that issue report is in the wrong repo.
>If you want you can try to boot via hekate. If it still doesn't work then reopen the issue.

closed 2021-07-09 by CTCaer

KikoEsUnDios, 2021-07-09:
> You are not using hekate to boot.
> As I said:
> 
> > Other than that you are not using hekate to boot, so that issue report is in the wrong repo.
> > If you want you can try to boot via hekate. If it still doesn't work then reopen the issue.

I know that you are not trying to be rude or anything, but if I'm not using Hekate to boot then how should I do it?

I thought that it was putting the fusee-primary.bin in the bootloader/payload folder, then on Hekate select Launch -> Atmosphere -> And that's it?. Is this not the proper way to do it?.

Btw, I'm not very familiar with GIT. Where should I post this issue then? In which repo?

impeeza, 2021-07-09:
@KikoEsUnDios this space is for reporting bugs on Hekate, your config is using hekate to boot plain atmosphére, and fot the little I read on this post you need is to learn a little of the support, and for that there is a plenty of sites,  you can start reading the guides and understanding the switch mod process.