CTCaer/hekate #635: I updated my switch on accident and now it is somewhat bricked

DrQuack102, 2021-07-17:
So I used hekate and everything worked fine, until I accidentally updated my switch, and can no longer boot the switch up, even under stock. I need some help as to what I should try to do next .Would reverting firmware help any? The specific error message I get is "Unknown pkg1 version, HOS version not supported! Failed to launch HOS!"

Yoti, 2021-07-17:
Update your hekate SD card files and send new hekate.bin payload to the system.

DrQuack102, 2021-07-17:
So I am still fairly new to modding in general, so how should I go about updating? Cause I saw that there was a update.bin file, but I don't know much about bin files, or do you mean redownload hekate? Because I already tried that, and still got the same message.

CTCaer, 2021-07-17:
use the button Logs ON at top right in Launch and post a image of the log after launching.

DrQuack102, 2021-07-17:
[image]

CTCaer, 2021-07-17:
you didn't upload the image.. You posted before the uploading finished

DrQuack102, 2021-07-18:
[image]

My bad, should be here now

CTCaer, 2021-07-18:
Update hekate.
Go to https://github.com/CTCaer/hekate/releases
get the release zip and follow the "How to update:" in big bold letters there.

closed 2021-07-18 by CTCaer

DrQuack102, 2021-07-19:
So my issue is that file, I can't seem to find it to inject it.

Ig that may have been better to start off with

CTCaer, 2021-07-19:
[image]
[image]

DrQuack102, 2021-07-19:
Thank you for helping me with my issue, and idk how I didn’t see that when I downloaded it the first time. But now I’m getting this issue after the logos pop up, and I just updated hekate, how would I go about updating sept? And the atmosphere as you said I should?
[image]

impeeza, 2021-07-19:
Please direct support question to discord, gbatemp or orhers sources of support, this site is for bug report. And clearly you have a serious case of no reading instructions  Your problem is easily solved reading the installation guide of Hekate and Atmosphère

brandonrc, 2022-11-19:
I had this same issue. 

> [image]
> 
> My bad, should be here now

Had the same issue. I just to the hekate menu at boot. Selected the emuMMC and the migration button. Then repaired the raw emuMMC and it started right up after that.

Switchfan12, 2023-10-13:
I tried this and now my entire screen is just a weird yellow color

impeeza, 2023-10-14:
closed issue from 2 years ago, don't think will help you.

any support should be looked for on respective sites, discord or GBATemp.