CTCaer/hekate #636: Pkg2 decryption failed!

VeoScool, 2021-07-17:
Hey I tried updating hetake but this happend, anyone know what I could do?

https://user-images.githubusercontent.com/31881448/126044442-070a2bf1-ae41-4665-8d28-af77387ab08b.mp4

impeeza, 2021-07-17:
You need to upgrade atmosphère and be sure of upgrading the injection file to last varsion of hekate payload

CTCaer, 2021-07-17:
If you are sure you also updated atmo (all folders/files it has) then post a screenshot/photo from Tools -> 2nd tab at the bottom -> Dump pkg1/2

VeoScool, 2021-07-17:
> You need to upgrade atmosphère and be sure of upgrading the injection file to last varsion of hekate payload

Oh okay, I now replaced/merged the new Atmosphère 0.19.5 files on to my sd card and I'm using the hekate_ctcaer_5.5.8.bin

It worked, thank you!

closed 2021-07-17 by VeoScool

CTCaer, 2021-07-17:
Basically that message can happen either 
- by having old Sept in fusee-secondary.bin (for hekate) or in /sept folder (for Nyx) and new pkg1/2 dictate a new mkey.
- Or corrupted pkg2 (normally happens on emuMMC with fake cards).
- Or if mkey identified for pkg1 is new but there's old pkg2 (so basically mismatched pkg1/pkg2 combo because of failed update or incomplete restore).