CTCaer/hekate #637: Nothing shows after booting sysmmc from HEKATE

karama89, 2021-07-18:
Hello and greetings
I dont know if this is the right place to ask or not.

I updated my firmware on sysNAND to 12.1.0 and my Atmosphere to 0.19.5 and my hekate to 5.5.8

now whenever I lunch to CFW sysMMC it goes to black screen only.

did I miss anything ?

Here is the log
[image]

CTCaer, 2021-07-18:
More info is needed. Also that's not the whole log. Try to catch it via video and then screenshot the video.
Btw you have 1bit sd card issue?

karama89, 2021-07-18:
thanks for your reply.
That is exactly what I did... took ta video and screenshot it.
what other info you need?

and yes.. i do have 1bit SD card issue. which i cant fix, because the internal card reader is kinda damage  so i copy the games to the internal memory.

CTCaer, 2021-07-18:
Well 12.1.0 has a new mkey and needs to run sept that doesn't support 1bit mode, reason why you get black screen.
So I guess the only thing you can do is to downgrade (boot0/1 and rawnand). And more importantly you need to be careful so you don't overwrite your MBR in your SD card where the keys live (I would suggest to mount sd on pc and backip that sector also).

You should really go to a repair shop and get that sd card reader connector on the motherboard fixed.

closed 2021-07-18 by CTCaer