CTCaer/hekate #638: PKG2 Decryption failed on Stock?

Aida-Enna, 2021-07-23:
Note: I'm not 100% sure on the technical terms of things, but I'll do my best. Sorry!

I dug my switch out to play Pokemon Unite. I set it up a while ago with an EmuNAND and SysNAND, as I think they're called? Basically, I can boot to stock and go online to play games I own, or boot into CFW (Atmosphere) to play ROMS and use homebrew. Switching back and forth between them via Hekate worked fine.

I upgraded the firmware on my EmuNAND using Daybreak, and updated Atmosphere. No issues booting into EmuNAND.

Today I booted into stock, upgraded the firmware via the HOS updater and played a bunch of Pokemon Unite. Awesome! But then I tried to boot into hekate and I noticed autorcm had been turned off. I'm not worried about downgrading or burning fuses or whatever, so I just got back into RCM via the jig, turned on AutoRCM via Hekate, then booted into EmuNAND. Everything seemed to be working fine. However, when I turned off my system and tried to boot into Stock via Hekate, it gives me the following error (with logging on):

```Initializing...

Found pkg1 (REDACTED).

Identified pkg1 and mkey 11

Found FSS0, Atmosphere 0.18.1-1545FA9D
Max HOS supported:L 11.0.1
Unpacking and loading components... Done!
Loaded config, pkg1, and keyblob
Generated keys
Loaded warmboot and secmon
Read pkg2
pkg2 decryption failed!
Is hekate or Sept updated?

Failed to launch HOS!

Press any key...```

Also, my understanding of how this works is that it shouldn't be booting into Atmosphere CFW at all, but into the Stock HOS. Am I doing something wrong? I downloaded the latest atmosphere and extracted it to the root of my SD card, and booted with the latest Hekate as well.

CTCaer, 2021-07-23:
Duplicate of https://github.com/CTCaer/hekate/issues/636

No it doesn't boot cfw when stock=1 is enabled.
stock=1 is exactly there so it can disable all cfw elements.

12.1.0 changed master key so new sept must run.
sept is from atmo, so you need to update atmo also.

closed 2021-07-23 by CTCaer

Aida-Enna, 2021-07-23:
I actually checked on that before posting - They were trying to run CFW and I stock, so I figured it was different enough.

And I'm not sure what to say - You can see in my hekate config, ``stock=1`` is there but it still tried to boot atmosphere as it shows in the logs.

CTCaer, 2021-07-23:
>it still tried to boot atmosphere as it shows in the logs

It doesn't. You are confusing the parsing of ffs0 to get sept and secmon with booting atmosphere.

>I have the latest Atmosphere already on the card

Your log says Atmosphere 0.18.1-1545FA9D which is very old.
So get a new log (take a photo, no need to write it down) **from both stock and emu cfw** and post them here.
Also make sure you are using the official atmo.

Aida-Enna, 2021-07-23:
>It doesn't. You are confusing the parsing of ffs0 to get sept and secmon with booting atmosphere.

Ah, okay.

>Your log says Atmosphere 0.18.1-1545FA9D which is very old.
So get a new log (take a photo, no need to write it down) from both stock and emu cfw and post them here.
Also make sure you are using the official atmo.
EDIT: also don't redact Found pkg1 again which has the pkg1 build version. The whole log has 0 personal info.

Ah, weird. I just deleted the atmosphere and sept folders on the root and re-extracted them from the ``atmosphere-0.19.5-master-9df13781c+hbl-2.4.1+hbmenu-3.4.0.zip`` archive, which should be the latest. The same error message pops up on stock, claiming the old atmosphere version. I thought the pkg1 thing was a key or something, that's why I took it out, sorry.

Here is the log from stock. I am not sure how to get the log for emu CFW because it boots correctly with no error messages.
[image]

CTCaer, 2021-07-23:
Yeah, just saw that `fss0=bootloader/payloads/fusee-secondary.bin`...
Why even, especially if you use CFW and not only stock.

**Fix:**
Change that so it points to the correct path (`fss0=atmosphere/fusee-secondary.bin`) and clean your payloads folder from stuff that are not "RCM payloads".

**Valuable info/explanation:**
What you did is not exactly wrong but it's normally only used by users that only want stock mode and no CFW.
(aka copying only fusee-secondary.bin and in a random place, in order to reduce clutter from a full atmosphere file structure).
And these users normally know that they only need to update that file when they update OFW/Stock.

For users that boot stock **and** cfw, that's completely needless and it can also make you forget to update it there.
In your case for some reason you changed that and forgot it completely.
And thus it was pointing in the very old one in /bootloader/payloads.

If you want to learn how to utilize hekate's config system properly you can check:
https://github.com/CTCaer/hekate/blob/master/README.md for info about ini keys and conditions/limits.
https://github.com/CTCaer/hekate/blob/master/res/hekate_ipl_template.ini for various boot entry examples

Ofc the template doesn't have everything you can do but it's good enough imo.
Also, even if here's not a support forum, I'm always happy to explain config .ini questions.

Aida-Enna, 2021-07-23:
Weird, maybe I goofed a copy/paste at some point. Either way, that fixed it. I'll check out that documentation as well - Thank you very much!