CTCaer/hekate #641: [Feature] Option in hekate_ipl.ini to autoboot into hekate — without pressing the Vol Down button

JrRadiant, 2021-07-29:
Hello!

One of my friend's Switch has damaged Vol - button, but he loves Hekate a lot, and I guess it will be super nice to have an option in hekate_ipl.ini to allow the user to autoboot into the Hekate **without** holding down the **Vol Down** button.

Is it possible from your POV?

Thanks in advance!

JrRadiant, 2021-07-29:
autoboot=0 option cured my friend's Switch, thank you & sorry for disturbing, hehe.:)

closed 2021-07-29 by JrRadiant

CTCaer, 2021-07-29:
Yeah, I was gonna say. You can control all options in case you can't enter the menu by editing hekate_ipl.ini or nyx.ini.

If your friend still wants to autoboot, hekate can be compiled and this line can be changed:
https://github.com/CTCaer/hekate/blob/master/bootloader/main.c#L1057
`BTN_VOL_DOWN` -> `BTN_VOL_UP`.