CTCaer/hekate #642: Failed to mount SD

Krstox, 2021-07-29:
Hi i formated my SD Card to Fat32 and now i get this error message:

"Failed to mount SD!"
"Missing LP0 (sleep Mode) libè
"Missing or old Minerva lib!

"Update Bootloader folder!"

What can i do, with the exfat format it recognize the card but atmoshpere doesnt work.

Masamune3210, 2021-07-29:
make sure card is using MBR partition style, make sure SD reader is secure, try unplugging and plugging it back in, possible SD card failure?

CTCaer, 2021-07-29:
You do some weird partitioning/formatting if you have that issue after setting it to fat32.

Dunno why you struggle with that custom formatter you use though, when you can just have it as exFAT and then format via hekate (Tools -> Partition SD Card). You can choose partition and sizes there and the fat partition is always made as FAT32

closed 2021-07-29 by CTCaer