CTCaer/hekate #644: [Feature request] USB 3.0 support?

ca1e, 2021-07-31:
I#s it possible that USB Mass Storage support 3.0 speed?

impeeza, 2021-07-31:
It does support 3.0 speed you need to read the instructions and apply the reg  file to you computer, please read the initial page

masagrator, 2021-07-31:
> It does support 3.0 speed you need to read the instructions and apply the reg file to you computer, please read the initial page

USB 3.0 setting is for Atmosphere. Not for Hekate itself which author is referring to.

CTCaer, 2021-08-01:
> Is it possible that USB Mass Storage support 3.0 speed?

Eventually I will add superspeed support to UMS.
Especially now that I mitigated the issue with ASMedia usb controllers on Linux/Mac OS internally.

impeeza, 2021-08-02:
I just swear what UMS was NYX feature, and the reg file on releases was to get the full speed, for me, using Nyx I can make a full emuNand partition backup in 17 minutes, and using a USB 3.1 card reader took only 10 secs less.

masagrator, 2021-08-02:
> I just swear what UMS was NYX feature, and the reg file on releases was to get the full speed, for me, using Nyx I can make a full emuNand partition backup in 17 minutes, and using a USB 3.1 card reader took only 10 secs less.

29.1 GB in 17 min is ~29MB/s which is well within USB 2.0 spec speeds.

CTCaer, 2021-08-05:
Well with a good sd you get ~39-41MB/s reads on ums. The issue is writes that become 2/3. But it's still a good speed.
The only reason I consider solid and useful for having superspeed support is for users that want to do full SD card backup/restores.
There it plays a big a role.

rangermix, 2023-05-16:
+1 for this feature. I'm getting steady 26MB/s for whole card backup and 60+GB takes 1 hour. Unfortunately, I don't have an sd card reader around.

paulober, 2023-05-20:
Isn't the USB-C port on the switch implemented (from a hardware perspective) so it CAN only run at USB 2.0 speed even thought the connector it self is capable of USB 3.0?

impeeza, 2023-05-20:
Connectors are nothing more than a physical connection media, they do not allow or deny speed (mostly)  by example USB 2.0 can use traditional USB-A connector and you can use full speed or normal speed.

since USB 3.0 there is a new set of connectors (some of them retro compatible) which introduces extra pins needed to achieve new protocols which helps on the new set of speed.  by ultimate is the USB chip control and his software the one responsible for the negotiation and use of the correct speed.

big N have a "Propietary" USB implementation but by the latest firmwares is able to use USB 3.0 speeds.

paulober, 2023-05-20:
@impeeza Yea but i mean how the connector is wired up to the internals and the controller restrict its speed

impeeza, 2023-05-20:
Ok, that can be true but on the Switch implementation all the signals of the USB-C are connected to the driver chip, so no restriction at all, you can get confused because almost hardware mod concentrate just on the D+ & D- signal and ignore the rest because they are useless for mods.  but the others signals are used by the driver and the TV output signals.

Masamune3210, 2023-05-20:
If the controller or the port were restricting the speed to 2.0, HOS wouldn't be able to use 3.0 either.

leonpano2006, 2023-06-11:
And Bluetooth and Wi-Fi 2.4GHz is not a issue in hekate (since hekate doesn't support wifi and Bluetooth in first place and is not needed)

mrgarbagegamer, 2025-02-20:
> > Is it possible that USB Mass Storage support 3.0 speed?
> 
> Eventually I will add superspeed support to UMS. Especially now that I mitigated the issue with ASMedia usb controllers on Linux/Mac OS internally.

I don't want to be the "*eta wen*" guy but I am curious, is this still on the roadmap? It's definitely something that would be useful, though I could understand other things being higher in priority

CTCaer, 2025-04-29:
It's in the roadmap of "If I get bored enough, I will port a PCIE driver and add USB3 support to the usb device driver" basically.

zoeleu, 2026-11-02:
Heya! I decided to implement this, and I have a working build. I used agentic AI to assist in the writing of this module, and I'm wondering what @CTCaer's policy on AI generated code is. I review, understand, and modify the code written by the AI models, and a lot of the code is just a port of the Linux driver. Would you be open to receiving a PR? There is one catch though, I only have a v1 switch to test with, so it's really unverified on every other switch.

zoeleu, 2026-11-02:
https://github.com/zoeleu/hekate/tree/usb3-ums

impeeza, 2026-11-03:
> 19\. Heya! I decided to implement this, and I have a working build. I used agentic AI to assist in the writing of this module, and I'm wondering what [@CTCaer](https://github.com/CTCaer)'s policy on AI generated code is. I review, understand, and modify the code written by the AI models, and a lot of the code is just a port of the Linux driver. Would you be open to receiving a PR? There is one catch though, I only have a v1 switch to test with, so it's really unverified on every other switch.

No IA slop code is accepted, if you write the the PCIE and USB3 drivers with quality human code it will be added.

zoeleu, 2026-11-03:
> No IA slop code is accepted, if you write the the PCIE and USB3 drivers with quality human code it will be added.

I am willing to attempt a complete rewrite without AI if a maintainer of Hekate confirms me that it is needed.

My commit is not large at all, about ~500 SLOC and a lot of it is essentially a direct port of the USB 3 driver from Linux, swapping out the device trees, clocks, regulators, etc, with the Hekate counterparts, or direct access to registers.

I triple checked the code for conciseness and it appears to be OK, but I am not near an expert in driver implementations or the USB 3 protocol, so I would appreciate some guidance from someone who knows more than me on the subject.