CTCaer/hekate #645: Enhancement suggestions - reload hekate from nyx without external injection &

zemixe, 2021-08-02:
If the user updates hekate files over UMS then hekate needs to be injected from RCM as the reload function only reloads nyx.

Two user friendly solutions I thought of:

1. Include a copy of the latest hekate payload at `bootloader/update.bin` in the release package, then update the reload function perform an injection.

2. Include a copy of the latest hekate payload at `bootloader/update.bin` in the release package, then in addition to `bootloader/payloads/` have the payloads menu check for update.bin so the user can reinject hekate from there.

Yoti, 2021-08-03:
But you may just move hekate_ctcaer_x.y.z.bin from the root to the payloads dir right after unpacking zip.

zemixe, 2021-08-03:
> 
> 
> But you may just move hekate_ctcaer_x.y.z.bin from the root to the payloads dir right after unpacking zip.

The idea is that the latest hekate payload gets copied via USB in UMS mode by *all users* since most likely don't do this or are even aware of what update.bin is. That in turn would ensure that the proposed reload function or a payload menu immediately have the latest hekate payload available, without reinjection. There's no real downside to including a copy there other than slightly increasing the release zip size.

Anyway this issue was a reminder requested by CTCaer who I'm sure will decide on the best way to do it.

CTCaer, 2021-08-05:
Nyx reload always run hekate main. And hekate always checks update.bin.
The trick for a UMS or HOS based update is to update /bootloader/update.bin and then do a nyx reload to get updated hekate and nyx. And launching a fss0 boot entry will also update reboot_payload.bin if it's hekate or force it as hekate if update r2p is enabled.

Anyway, I will consider including update.bin as is.
For now, the reason I don't include update.bin is because I want latest version enforced.
If I include it and a user pastes an old version and injects that it will run that, instead of getting the dragons message because of newer hekate and older nyx.

Anyway, for now I will include a how to message in releases on the how to update though, to ease your mind.

zemixe, 2021-08-05:
Ok understood, I didn't realize that. You confused me because you originally said: 

"have in mind that reload nyx only reloads to new Nyx, so it's best to reboot into rcm and inject new hekate after copying the files" without mentioning that adding the latest hekate at update.bin would also fix it :)

CTCaer, 2021-08-05:
Reload Nyx basically chainloads hekate binary saved in ram.
hekate always checks update.bin on start. So effectively with reload, you can say you also force a recheck for update.bin.

>"have in mind that reload nyx only reloads to new Nyx, so it's best to reboot into rcm and inject new hekate after copying the files"

Forgot that this was from the convo in discord.
I basically meant that Nyx on its own doesn't read anything from storage while pressing that button.
hekate copies itself in ram and Nyx gets it from there.
Also by "make a suggestion issue in github so I don't forget and I will think about it" I meant make a suggestion for update.bin to be in release zip.

Anyway, my stance remains the same with what I said in here and in discord. I will think about it.

CTCaer, 2021-09-19:
v5.6.1 now has update.bin pre-bundled.

closed 2021-09-19 by CTCaer