CTCaer/hekate #648: Game Card won't load

km159851, 2021-08-20:
Yup, it's one of these again. Sorry.

##What's the issue you encountered?

Game card error. Neither PGLE nor Sword would load on Atmosphere CFW. Both load on stock just fine.

##How can the issue be reproduced?

Launching CFW and trying to use a game card. Doesn't matter if I launch and then insert or launch with it already inserted.

##System Firmware Version

Switch: 12.1.0
Hekate: 5.5.8 https://github.com/CTCaer/Hekate/releases/
Nyx: v1.0.5 https://github.com/CTCaer/Hekate/releases/
Atmosphère: 0.19.5 (https://github.com/Atmosphere-NX/Atmosphere/releases)

##Additional context?
autonogc=0, nogc=0 in BCT.ini, and there is no loader.kip

apparently being a 10 pin RCM switch is relevant too? not sure why but okay.

masagrator, 2021-08-20:
Hekate has separate option for nogc that is overriding Atmosphere's one.

CTCaer, 2021-08-20:
If your stock is 12.0.2 and up and your emummc is older, you can't use GC because GC fuses are burnt.
If that's the case, update your emummc.

km159851, 2021-08-20:
> Hekate has separate option for nogc that is overriding Atmosphere's one.

Yes, and I turned that one on too. They are both on.

masagrator, 2021-08-20:
> > Hekate has separate option for nogc that is overriding Atmosphere's one.
> 
> Yes, and I turned that one on too. They are both on.

They must be turned off (0), not on (1)

km159851, 2021-08-20:
> They must be turned off (0), not on (1)

Sorry, yes, they're both zero, I just woke up and my brain is dead.

km159851, 2021-08-20:
> If your stock is 12.0.2 and up and your emummc is older, you can't use GC because GC fuses are burnt.
> If that's the case, update your emummc.

That worked! Thank you, I didn't know I needed to do that.

closed 2021-08-20 by km159851