CTCaer/hekate #649: [BUG] cannot show the last boot option

carcaschoi, 2021-08-23:
in the launch menu
[image]

「Hack sysnand」 word doesn't show up

masagrator, 2021-08-23:
Upload here hekate_ipl.ini

carcaschoi, 2021-08-23:
> Upload here hekate_ipl.ini

This is the bug of hekate. I know what I am doing.

carcaschoi, 2021-08-23:
My hekate_ipl.ini config:

[config]
autoboot=0
autoboot_list=0
autohosoff=1
autonogc=1
backlight=100
bootwait=5
updater2p=1

{ShallowSea(CFW packs for atmosphere and sxos)}
{Github: https://github.com/carcaschoi/ShallowSea}

[CFW Ams (fss0)]
fss0=atmosphere/fusee-secondary.bin
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/atmosphere.bmp
kip1patch=nosigchk
cal0blank=1

[CFW Ams (fusee-primary)]
payload=bootloader/payloads/fusee-primary.bin
icon=bootloader/res/atmosphere.bmp

[Stock OFW (No CFW)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
icon=bootloader/res/icon_stock.bmp
stock=1

[Hekate menu]
icon=bootloader/res/icon_hekate.bmp
payload=/bootloader/update.bin

[CommonProblemResolver]
icon=bootloader/res/icon_payload.bmp
payload=bootloader/payloads/CommonProblemResolver.bin

[Lockpick RCM]
icon=bootloader/res/icon_lockpick.bmp
payload=bootloader/payloads/Lockpick_RCM.bin

[TegraExplorer]
payload=bootloader/payloads/TegraExplorer.bin
icon=bootloader/res/icon_payload.bmp

[Hack sysnand]
fss0=atmosphere/fusee-secondary.bin
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/atmosphere-sys.bmp
atmosphere=1
emummc_force_disable=1

carcaschoi, 2021-08-23:
studio pancake overlays can show the config. Hekate cannot show the name of it but the current launch function is still work.

Sorry for my poor English

Masamune3210, 2021-08-23:
> > Upload here hekate_ipl.ini
> 
> This is the bug of hekate. I know what I am doing.

Should probably not be super aggressive to people you are asking for help from or reporting an issue to.....

CTCaer, 2021-08-23:
Bug is on the label

carcaschoi, 2021-08-23:
> > > Upload here hekate_ipl.ini
> > 
> > 
> > This is the bug of hekate. I know what I am doing.
> 
> You know what you are doing, but in first post you have used japanese quotation, while in ini file it's not used. You clearly don't understand possible implications of that, so stating that you know what you are doing in this case is baffling.

I am sorry for my misunderstanding. I misread it as 「Update here hekate_ipl.ini」lol😣

CTCaer, 2021-08-28:
Fixed in latest release

closed 2021-08-28 by CTCaer

DraconicNEO, 2021-10-24:
> 
> 
> in the launch menu [image]
> 
> 「Hack sysnand」 word doesn't show up

Sorry if this is off topic but where did you get the icon for the sys cfw?