CTCaer/hekate #651: Second boot "failed to load thk.bin"

p928s1984, 2021-08-29:
Not sure why. first boot went great. rebooted, and now I get an error stating "failed to load thk.bin". 
I have double checked the release package and it is not included. 
I can see in the code where it is getting this error, but no idea what it means. "// Use custom TSEC Hovi Keygen firmware."
https://github.com/CTCaer/hekate/blob/197ed8c319bd4132e4d7571ce037d4a27f806bba/bootloader/hos/hos.c#L416

hopefully its just a simple mistake on my part. 

12.1.0
running latest release (5.6.0)

p928s1984, 2021-08-29:
Update, recopied the bootloader folder. it re ran the scripts and seems to be happy again. must have been some sort of corruption.

CTCaer, 2021-08-29:
You probably checked the wrong thing the first time because it's included.
But for anyone's interest, that error happens when that file is missing or its folder or the file is corrupted and can't be read.
So re-extracting the release zip's bootloader folder is the way to fix it.

closed 2021-08-29 by CTCaer

p928s1984, 2021-08-29:
Not sure what you mean by checked? That file does not exist in the release zip... 
even now that it is booting, that file still does not exist...

I did not change any settings after first boot, just rebooted the console to verify all was working well...

CTCaer, 2021-08-29:
It's in `/bootloader/sys/` so `/bootloader/sys/thk.bin`. Not in sd card root `/thk.bin`.
You even read the code...
Same thing with the release zip.
It's inside sys folder that is inside bootloader folder. Yes these 2 (bootloader and sys) are folders, not files.
You can open them and they will show more stuff.

p928s1984, 2021-08-29:
100% my fault. I somehow downloaded someone elses fork/verison, which does not contain it. that would explain why it is erroring.

impeeza, 2021-08-31:
I am positively sure the file is on it:

[image]