CTCaer/hekate #652: Switch V1 Erista erro  BCT  ao tentar atualizar para versao 12.1.0 atmosfere

AdeltonAltair, 2021-09-01:
Boa noite  a todos , tenho um switch  v1 , instalei o chip sxcore modo chip  , na versao 10.0.4 resolvi atualiza para versao 12.1.0 para rodar jogos mais atual porém não deu muito certo sempre ia fazer o tutorial apresentava este erro "Failed to run sept Main BCT is Improper! Run sept with proper BCT at least once to cache keys."Resolvi atualizar utilizando ChoiDuoju então  deu certo porem carregou a tela do sxos e ficou preta , então pensei que tinha dado brink ,para ter certeza resolvi remover o chip de desbloquei somente ele não removi as soldas do processador grafico , voltei a nand para seu local original então liguei e funcionou porém sem desbloquei pois au tentar desbloquear via software da o mesmo erro de BCT, alguem sabe como corrigir desde ja agradeço.
[image]

AdeltonAltair, 2021-09-01:
De hecho amigo, seguí un tutorial de un canal, pero no me había dado cuenta porque era para Mariko v2 no para Erista pero creo que el modo chip ha cambiado algo en el arranque, porque antes de actualizar a 12.1.0 al poner los archivos ya dio el error.

Na realidade amigo , segui um tutorial de um canal , mais não tinha me atentado pois o mesmo era para Mariko v2 nao para Erista mas acredito que o modo chip tenha alterado o algo no boot , pois antes de atualizar para 12.1.0 ao colocar os arquivos ja dava o erro.

AdeltonAltair, 2021-09-01:
Pior que agora  esta ok, mais sem desbloqueio não consigo achar um solucao

AdeltonAltair, 2021-09-01:
Quero voltar  com o sxcore mais rodar atmosphere 12.1.0.

Masamune3210, 2021-09-01:
SX Core is showing issues on the atmos issue tracker with respect to the new boot methods, are you using spacecraft-nx or the original firmware

AdeltonAltair, 2021-09-01:
Então na realidade não uso emunand, hj o switch está sem desbloqueio nenhum ,mais ainda de encontra com o Sxcore porém removido não totalmente , só retirei o chip onde vai encaixada a Nand , voltei ela no lugar original mais se quiser voltar o chip e fácil, detalhe ao colocar o chip o logo da sxos volta depois fica em tela preta

AdeltonAltair, 2021-09-01:
Do jeito que está agora abre normal mais sem desbloqueio e já atualizado para 12.1.0

CTCaer, 2021-09-01:
@AdeltonAltair 
Only english here. If you are not fluent you can use a translator.
As for your issue, update to hekate. Your are using an old version. Sept was removed.

closed 2021-09-01 by CTCaer

Ton2301, 2021-09-02:
Olá Mesmo  atualizando nao obtive sucesso  , sempre utilizei o sxos mais como agora  esta complicado resolvi utilizar  Atmosphere no modochip.

Masamune3210, 2021-09-02:
You were already told to use english or a translator, please do so. Most people here only speak English so if you want help that's the best way to start

Ton2301, 2021-09-02:
Hi , I did not understand , I apologize , I really do not speak English I know few words but I will try to translate correctly .

Masamune3210, 2021-09-02:
That's fine, if you have to use a translator like translate.google.com. May not be perfect but should help at least some

Ton2301, 2021-09-02:
So, I'm not very knowledgeable about unlocking, I'm a computer technician with knowledge in electronics, I even installed the chip mode myself, sxcore was working in version 10.0.4, but I decided to upgrade to 12.1, in the process that I tried to update it was the same error first I'll try to update the hekate and see what happens

Masamune3210, 2021-09-02:
Have you tried using spacecraft-nx? Gateways crappy boot code is known to mess up the boot environment enough to cause issues for atmosphere and anything using it's associated methods of boot. Spacecraft-nx should get around this as it's sane