CTCaer/hekate #653: Pkg2 decryption failed! Is Hekate update? Failed to launch HOS!

Arktur0s, 2021-09-05:
Hi everyone !
Got this error and Ive done a clean install of Hekate ... Can someone help me with this please ?

Thanks !

Masamune3210, 2021-09-05:
Either you failed to update hekate properly and are launching a old version, or you have some corruption in your and. Make sure you copied the latest bootloader folder to the sd card and are pushing the latest hekate .bin

impeeza, 2021-09-05:
This space is for issues or bug reports. Having said that, you are injecting an old version of payload. Update your injection method

Arktur0s, 2021-09-05:
I know that I am injecting the right hekate.bin ... Ive done a reinstall and I got the same error ... maybe its the hekate.ipl that I put in the bootloader folder ?

masagrator, 2021-09-05:
pkg2 decryption failing is not related to hekate_ipl

Arktur0s, 2021-09-05:
Ok then I dont really know why its hapenning because I download the last Hekate ... deleted the bootloader folder and copying the new one and injecting 5.6.0 and getting this error dont know what Im doing wrong.

impeeza, 2021-09-05:
Test your sd card with h2testw, verify is formated on FAT32.

Decompress the hekate.zip https://github.com/CTCaer/hekate/releases/download/v5.6.0/hekate_ctcaer_5.6.0_Nyx_1.0.6.zip on a temp folder BEFORE coping to SD.

You can backup bootloader folder, delete from SD card and copy the recent decompressed bootloader folder

Update your injector with hekate_ctcaer_5.6.0.bin, renamed payload.bin if necessary

Arktur0s, 2021-09-05:
Indeed my sd is formated in FAT32 ... I use hekate since the first release and its my first problem with it ... Ive done what you asked impeeza and I get the same error again and again ...

impeeza, 2021-09-05:
Switch model?
Fw version?
Emmumc?
Payload injection method?

impeeza, 2021-09-05:
Any time do you used another CWF besides Atmosphère?

Arktur0s, 2021-09-05:
Thanks for helping me ... Only AMS as CFW

CTCaer, 2021-09-06:
If you have the latest hekate (5.6.0) and you see that number in the menu and your fw is 12.1.0 then your pkg2 is wrongly flashed.
It's either corrupt or old (mismatched).
Use an unbrick guide to fix that (aka flash the correct one).

closed 2021-09-06 by CTCaer