CTCaer/hekate #656: [Feature request] Load in the boot entry a .txt file for Dns Mitm

Domyx-01, 2021-09-09:
I wonder if it is possible to add this functionality: could you add a function to use on the hekate_ipl.ini file that allows you to load a .txt file to use for dns mitm?
It would be useful in case if you have 2 Emunand installed in the SD card (one to use it online and the other to use Homebrews, mods, cheats safely), currently Dns Mitm can be used to block Nintendo addresses on SysMMC, EmuMMC or EmuMMC with emuMMC ID (Read Dns Mitm documentation = https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/features/dns_mitm.md). Currently with the FSS0 start of Hekate I cannot block the connection to Nintendo on one of the 2 Emunands but I am forced to block it on both because the emunands created with Hekate always have the ID 0x0000.
You could solve the problem by creating a similar function for Hekate's prodinfo blanking that can be applied to any boot entry you want plus you can tell it the path to the .txt file to load.

I leave a conceptual example below:

`
[Emunand 1 - online]
fss0=atmosphere/package3
emummcforce=1
emupath=emuMMC / RAW1

{}

[Emunand 2 - offline]
fss0=atmosphere / package3
emummcforce=1
emupath=emuMMC / RAW2
cal0blank=1
dnspath=atmosphere / hosts / emummc.txt

{}`

In the example above the Emunand 1 did not choose a path for Dns Mitm so it is disabled, only default.txt will be loaded.
While in Emunand 2 I chose a path so it will be loaded on Dns Mitm in addition to default.txt also emummc.txt.
This is just a conceptual example, feel free to modify the operation.

CTCaer, 2021-09-12:
Atmosphere wrongly uses the emummc id for that. That key was DOA and was added with no design planning behind it.
For example it was supposed to be 16bit and not 32bit.

On the other hand it's possible to repurpose it in a way that works as is. I will think about it.

CTCaer, 2021-09-19:
Added in new release.
Use change emummc for your emummc.ini to get the new values.

closed 2021-09-19 by CTCaer