CTCaer/hekate #658: 1 bit mode and cpu

Daigo1317, 2021-09-17:
hi guys, my sd is analyzed in 1 bit mode, so I brought it back to the shop for repair, but they said that one of the pins is moved but still makes contact and that instead the cpu is struggling to load the games, the problem of the cpu can really cause 1 bit mode?

Masamune3210, 2021-09-17:
No, the shop is on crack. There is almost less than 0% chance that the cpu is at fault. It's either the sd connector, the reader, or the sd card itself.

CTCaer, 2021-09-18:
Pins moved and cpu is struggling, as masamune said, is a failed attempt on an excuse.
Either the pin is on the air and it loses connection and that kills init or later reads that cause a reinit to 1bit, or it doesn't touch the sd reader pin.
In either case, the fix is the same. Remove the problematic pin and solder a tiny wire from testpads and bend it and place it on the pin position, or solder it to the sd reader.
(Best way is the former, because the later introduces signal noise and skew)

closed 2021-09-18 by CTCaer

Daigo1317, 2021-11-13:
they changed the connector but still have problems

drzzzf, 2021-11-14:
The pin of the sd card is at the upper left corner of the cpu. If there is a solder joint there will cause the sd card to be inaccessible or in low-speed mode.