CTCaer/hekate #660: Yellow Screen on boot

GoldenSloth3434, 2021-09-19:
I believe I've updated everything correctly to the latest versions that came out a few hours ago, but when I inject I get nothing but a [bright yellow screen](https://i.imgur.com/52byHGF.jpeg) Anyone else having this problem or know how to fix it?

CarlosRamirezGriego, 2021-09-20:
I have the same issue, I exported everything from the zip file in root

Regarding the Hekate version I used:
- Version 5.6.1
- Hekate bootloader files were also in root and replaced the ones from before

CarlosRamirezGriego, 2021-09-20:
The Yellow screen went away because I was missing 1 step.
The hekate documentation specifies that we need to edit "hekate_ipl.ini"
Replace "fss0=atmosphere/fusee-secondary.bin" for "fss0=atmosphere/package3"
This removed the yellow screen but resulted in another error for me

urherenow, 2021-09-20:
First thing you should do is launch fusee (vanilla Atmosphere) from nyx to see if it works. If it works, then something is wrong with hekate_ipl.ini

CTCaer, 2021-09-20:
Mismatching files for atmosphere can cause that. (Yellow screen or black screen/rcm or fatal error with 0100000000000008)
Editing `fss0=atmosphere/fusee-secondary.bin` to `fss0=atmosphere/package3` in `bootloader/hekate_ipl.ini` is mandatory if you update to atmosphere v1.0.0 and up.

closed 2021-09-20 by CTCaer

GoldenSloth3434, 2021-09-20:
Can someone share their hekate_ipl.ini file? I had mine, did the edit, and it worked, then misclicked and deleted the bootloader file half an hour later, and no fresh zip has the file for some reason