CTCaer/hekate #661: DNS Mitm with emummc id not work again

Domyx-01, 2021-09-19:
Hi CTCaer, I'm the guy from last time who opened this issue = https://github.com/CTCaer/hekate/issues/656.
I tried the latest version you released (version 5.6.1), but unfortunately even with the new id associated and created by emunand it doesn't work.
Then I had decided to open a problem also from the Atmosphere Github page and Sciresm told me to try to start the emunand through fusee.bin and incredible it worked through the new ids generated by Hekate, but for some absurd reason if I start through FSS0 not it works.
SciresM said these words:

"Sounds like hekate isn't setting the emummc_id field when booting exosphere."

I don't know if these words from him can help fix this bug.

I leave you the link of the problem that I had opened from Atmosphere to read our conversation: https://github.com/Atmosphere-NX/Atmosphere/issues/1628

Thanks again for everything you do, you are great and I love your new version, there is only this problem left to solve.

CTCaer, 2021-09-20:
@Domyx-01 try this one
[hekate.zip](https://github.com/CTCaer/hekate/files/7194106/hekate.zip)

CTCaer, 2021-09-20:
Update to v5.6.2 from releases. That fixes the issue and also now sets ID when you use `emupath=` key in a boot entry to change emuMMC on the fly.

closed 2021-09-20 by CTCaer