CTCaer/hekate #664: Change default hekate_ipl.ini

Syko7956, 2021-09-25:
Multiple people in modding circles have expressed trouble booting to anything (sysnand/emunand) recently. With atmosphere 1.1.1 and hekate v5.6.2 & Nyx v1.0.7, it IS possible for people to enjoy it on Firmware 13.0. But it's not explained that you have to change this file to prevent the lime green screen of death. 

Fuuse primary is no longer used. Those who crash had to change this in the ini:
fss0=atmosphere/fusee-secondary.bin into fss0=atmosphere/package3

urherenow, 2021-09-25:
This was written in the release notes. Close the issue, because it isn't one, and learn to RTFM.

masagrator, 2021-09-25:
```
Notice about Atmosphere 1.0.0 and up: The fss0=atmosphere/fusee-secondary.bin in hekate_ipl.ini must be changed to fss0=atmosphere/package3
Notice about resized emuMMC: Resized emuMMC does not have the slow bug since v5.6.0. You can create a new one or recreate it via Nyx and the issue will be gone.
```

Not explained, you say?

closed 2021-09-25 by Syko7956