CTCaer/hekate #665: It is possible to emulate reboot->OFW as a launch option?

bpcommunity, 2021-09-25:
I think that it's safer to use this option because I don't need to use Atmosphere or being locked out if `package3` does not support some fw upgrade.

It's similar to genuine boot, I guess. I read documentation, but I was not able to "emulate" that option (reboot->OFW) with configurations to `hekate_ipl.ini`.

Is that possible? (just to know if my research will reach a happy end).

impeeza, 2021-09-25:
Remove autoRCM or boot without RCM you will boot on genuine OFW, burn fuses and load everything from your MMC

bpcommunity, 2021-09-25:
> Remove autoRCM or boot without RCM you will boot on genuine OFW, burn fuses and load everything from your MMC

Yes, I know that, but I want to do the same as `hekate->reboot->OFW`, but as a launch option. If that's not possible, no problem. But if it is possible, I want to know what kind of conf I need to read about.

CTCaer, 2021-09-26:
It's possible to be implemented, although I don't find a use when you can press the button.
Especially in your case where it doesn't matter. Also even if it gets implemented, it will act like Reboot OFW option and it needs AutoRCM disabled or at least one BCT entry intact.

Only usage I can think of is for using an app/overlay to reboot to that option immediately from HOS.

bpcommunity, 2021-09-28:
Thanks for your answer @CTCaer. Sounds good enough for me. Gratitude.

closed 2021-09-28 by bpcommunity

antonwe, 2024-02-27:
I'd like to revisit this, with modchips in mind. On my switch with a modchip installed, I'd love to immediately be presented with the launch screen with OFW boot as one of the options. I can certainly bypass the modchip by holding volume buttons during power-on or press reboot->ofw in Hekate, but when I hand the switch over to someone who doesn't fully understand how things work, having OFW as a big nice launch button would be far more obvious.

Mrgrin3h, 2024-02-27:
Just don't give it to them, or start it before you give it to them

impeeza, 2024-02-28:
@antonwe you can edit your Hekate_ipl.ini file and disable autoboot, or USE NYX interface to do it (you can enter in to nyx pressing VOL-)

antonwe, 2024-03-29:
@CTCaer would you please reconsider this, I know plenty of users who would want this feature, simply for ease of usability purposes.

urherenow, 2024-03-29:
It’s already an option. RTFM.

CTCaer, 2024-03-29:
Just use Stock (`stock=1`)

antonwe, 2024-03-29:
> Just use Stock (`stock=1`)

I got the impression that this isn't the same thing as reboot->OFW. What is the difference?

theofficialgman, 2024-04-03:
@CTCaer I could be missing something but using `stock=1` still requires atmosphere's package3.
It would be desirable to be able to use hekate's ability to run OFW without needing atmosphere's package3.

I personally would find this useful as I have an OFW option for R2C but need to remember to also update atmosphere package3, even though I do not have atmosphere installed and do not use atmosphere.

urherenow, 2024-04-04:
You can already do that from within NYX

didix16, 2024-11-24:
Hi guys, I know this is closed but I just made a PR because I wanted this functionality too.

I'm totally agree with @antonwe.

> I'd like to revisit this, with modchips in mind. On my switch with a modchip installed, I'd love to immediately be presented with the launch screen with OFW boot as one of the options. I can [...]

So this is the PR #1044. Hope @CTCaer want it to implement.

Thanks in advance.