CTCaer/hekate #666: Hekate unable to start when EmuMMC is disabled

vproper, 2021-09-25:
When I disable EmuMMC in Hekate EmuMMC settings, it corrupts the emummc.ini file, so Hekate is unable to start properly and it shows only black screen. Erista unpatched model

BestPig, 2021-09-25:
Oh, I got the same issue yesterday.
I didn't remember if I was in 5.6.1 or 5.6.2.
I used a File Based emummc on an Erista unpatched if it can helps.

[image]

DavidBuchanan314, 2021-09-25:
Anecdotally, two instances of this exact same emummc.ini corruption have cropped up recently in the reswitched user support channels. I don't have the specifics, but I thought I'd add this as evidence towards it *not* being random FS corruption unrelated to hekate.

DavidBuchanan314, 2021-09-25:
A total stab in the dark as to what the cause could be, but on this line `path` could be null, which might make strlen unhappy:

https://github.com/CTCaer/hekate/blob/af7bee2231560a708129e7f417c7f0fd036d7692/nyx/nyx_gui/frontend/fe_emummc_tools.c#L116

vproper, 2021-09-26:
> A total stab in the dark as to what the cause could be, but on this line `path` could be null, which might make strlen unhappy:
> 
> https://github.com/CTCaer/hekate/blob/af7bee2231560a708129e7f417c7f0fd036d7692/nyx/nyx_gui/frontend/fe_emummc_tools.c#L116

So, I think what we need is to check if the string is null before proceeding to strlen

urherenow, 2021-09-26:
0 is not null. It's already set to 0.

CTCaer, 2021-09-26:
Yes this should check if path is NULL.
The issue can only happen when disabling emuMMC. And it produces a garbage path, suffixed with "/Nintendo" string.
In the end, what happens from all that is a heap corruption because emummc path max size is 128 chars.

Thing is that I wonder why strlen finds data there.
Normally BPMP sees zeros for the first bytes in 0 address. (Yes, BPMP doesn't abort reading there.)

DavidBuchanan314, 2021-09-26:
is address 0 writable? Perhaps a bug elsewhere is causing data to be written to NULL.

CTCaer, 2021-09-27:
It's not writable. Reason why I wonder.
But anyway, I will push a release tomorrow.

rashevskyv, 2021-09-30:
Any progress?

Viperbite, 2021-09-30:
Where did id=0x8eaff come from?? Maybe it doesn't matter, but in my old emummc.ini it looks like this...
[image]

Masamune3210, 2021-10-01:
If its what I think it is, going into nyx and updating your emummc settings will generate the id, its used to identify that specific emummc instance. If I'm not, ignore me

CTCaer, 2021-10-01:
@Viperbite it's normal in newer versions. Before, the id was always 0 because it was obsolete and not used. Now it gets autogenerated from path.

As for progress on path issue; the actual resulted issue on that is that compiler optimizes out later path checks because path was read once without checking. So I needed to find time to make sure that this doesn't happen elsewhere.
Maybe today, I'll do a release.

CTCaer, 2021-10-01:
Fixed in latest release (v5.6.3/1.0.8)

closed 2021-10-01 by CTCaer