CTCaer/hekate #667: It is possible to restore emummc into sysnand?

davedave85, 2021-09-26:
hi,

i want to to know if its possible to restore emummc into sysnand and if the procedure is the same to restore any backup?

impeeza, 2021-09-26:
Hi, this is not an issue.  This space is for report bugs.  

The answer to your question is on the main page ot nyx section. There states how to restore a backup, if you use a hidden partition you must make a backup to files of it. Then restore that backup on the sysnand. You can use nyx or emutool for the backup.

davedave85, 2021-09-26:
> Hi, this is not an issue. This space is for report bugs.
>
> The answer to your question is on the main page ot nyx section. There states how to restore a backup, if you use a hidden partition you must [...]

thanks and sorry

closed 2021-09-26 by davedave85

impeeza, 2021-09-26:
Worry not 😀 you can get help on discord or gbatemp or another forum you like

davedave85, 2021-09-26:
oh its possible to share the discord? 😀

impeeza, 2021-09-26:
I don't know which because I don't use discord but I think can be found using Google 👍