CTCaer/hekate #677: [Enhancements] Sleep Mode and Screen Lock in Hekate

DraconicNEO, 2021-11-04:
I think that it would be a good idea to add a Sleep mode to Nyx/Hekate to turn off the screen by pressing the power button as well as a Screen lock function similar to the touch based TWRP Android Recovery mode.

[image]
Example of the Screen Lock Function in TWRP

The screen lock feature isn't for security, it just ensures that buttons don't get randomly pressed if it's in a Pocket/Bag and gets woken from sleep mode on accident.

Masamune3210, 2021-11-04:
What would be the purpose of this? Hekate isn't designed to really be stayed in unless you have something you are doing in it, if you aren't why not just boot whatever you are going to boot and then use it's facilities for screen lock

EmreTech, 2021-11-08:
Hekate is a bootloader. It's meant for you to launch a payload or boot another OS, not to stay on it.

EmreTech, 2021-11-08:
For example, you don't put your computer to sleep when it prompts you which drive you want to boot from.

CTCaer, 2021-11-08:
The tally on arguments for now is 1 - 0.5 (in favor of DraconicNEO )
(I would put +1 for Masamune3210, but I will leave to 0.5, just because I have a personal usecase that drops that argument)

Mind that arguments should be proper.

(Also have in mind that hekate if enters sleep mode it can stay there for **at least 3.5 weeks** before 0%, like l4t linux for switch. But some do not use linux.)
(And lastly that switch is special, where you need to exploit rcm and inject a payload on v1)

EmreTech, 2021-11-08:
Why are you tallying arguments? Seems random to me

Masamune3210, 2021-11-08:
Wasn't meaning to sound malicious. If I came off that way I apologize

impeeza, 2021-11-09:
I am not pro or con, but sometimes i.e. when you are using Nyx on long time consuming task like making a NAND backup it sound logic the lock screen so you don't abort the process by error, or your loved other interaction:  "honey what are you doooiiing..." (tap screen and abort on 99%) 😆

EmreTech, 2021-11-09:
Hmmmmm... you do have a point. Some tasks performed by Hekate take a long time, and in order to not interrupt the current task, a sleep mode could be added

EmreTech, 2021-11-09:
So here are some more counter-arguments to this:
Why would you tap the screen when someone needs your attention? Seems pretty random.
And when you stop a NAND Backup, can't you just resume it afterwards?
Also, why would file operations be performed while in sleep mode? Sleep mode means the device is sleeping, and is performing minimal operations.
Finally, why can't you just plug in your Switch and leave it idle while tasks are being performed? Most of the time, when these operations are performed, the user has a lot of free time.

> I am not pro or con, but sometimes i.e. when you are using Nyx on long time consuming task like making a NAND backup it sound logic the lock screen so you don't abort the process by error, or your loved other interaction: "honey what are you doooiiing..." (tap screen and abort on 99%) 😆

Masamune3210, 2021-11-09:
One reason just thought of would be with the OLED models, still images are bad even if dimmed if left long enough, maybe instead of sleep, screen off timeout if not already a thing?

CTCaer, 2021-11-09:
@EmreTech It's extremely simple on why I'm tallying them. I always do that when someone asks me to implement a feature and I don't want or find it important to use my free time to implement it properly. Especially for such a feature which will need extra work on mariko (use stock warmboot and make it run a CCPLEX payload that will then reset BPMP, run hekate and deinit CCPLEX). Always did that tallying but not public. And because a counter answer was demoralizing to OP, I reminded here that arguments should be proper and with thought.

@impeeza you can't abort such operations. I made it that way on purpose. Only VOL UP + VOL DOWN can abort them.

@Masamune3210 +1 on lockscreen reasons. (Although with a good sd, backup/restore only takes 11min so dunno. But in case of user forgetting it there, it's a solid +1)

(BTW all should consider sleep and lockscreen 2 different things, because they are used on and do different things)

EmreTech, 2021-11-09:
Alright, thanks for clarifying

DraconicNEO, 2021-11-10:
> 
> 
> Why would you tap the screen when someone needs your attention? Seems pretty random.
>

@EmreTech You probably wouldn't randomly tap the screen but if you were to put it in your bag or in a pocket the screen could be randomly clicked.  It wouldn't be able to do anything while the Nand backup/restore is in progress but it could after it's over.
Also another cause for this issue would be if you

>
>Finally, why can't you just plug in your Switch and leave it idle while tasks are being performed? Most of the time, when these operations are performed, the user has a lot of free time.
>

Not always, it really depends on where you are, remember the switch is a portable device so these operations may be done in a situation where that isn't an option and in that case you'll probably just put it in your bag or your pocket where it has the chance of having the screen triggered randomly.

DraconicNEO, 2021-11-10:
>
> (BTW all should consider sleep and lockscreen 2 different things, because they are used on and do different things)

I changed the issue to be only about locking the screen since it seems a bit more important and useful than Sleep mode.

bfenty, 2021-11-16:
While I don't think I'd normally use it, a lock could be useful to keep from accidentally hitting something during a long operation (such as format or NAND backup).

EmreTech, 2021-11-17:
> While I don't think I'd normally use it, a lock could be useful to keep from accidentally hitting something during a long operation (such as format or NAND backup).

A bunch of other people definitely agree with this. I might agree with this, but it might have a little use case. Especially since it's a little harder to stop long tasks anyways. CTCaer planned ahead by making it so you have to hold down two keys so you won't accidentally stop it. 
I also see how maybe something else could press down those keys, yet remove the case for that. But that is unlikely to happen.

DraconicNEO, 2021-11-17:
> A bunch of other people definitely agree with this. I might agree with this, but it might have a little use case. Especially since it's a little harder to stop long tasks anyways. CTCaer planned [...]

Even so, once the Nand backup is finished something could still press buttons on the screen. if someone put the switch into their bag and forgets they might not pull it out again until after the Nand backup has finished.

leonpano2006, 2023-06-11:
Can this with auto dark screen on switch? (specially OLED/AMOLED) (to prevent if placed too long and get burned)

Masamune3210, 2023-06-11:
Unless you are staying in hekate for literally month long stretches at a time, you arent going to get burn in lol

leonpano2006, 2023-06-11:
Maybe for an hour
And if this happen for 30times or more
(Just fail safe)

DraconicNEO, 2023-06-12:
> Can this with auto dark screen on switch? (specially OLED/AMOLED) (to prevent if placed too long and get burned)

I think the screen locking feature and a screen dimming function should probably be handled as separate feature requests.