CTCaer/hekate #678: old nyx gui found there will be dragons

Marshmellin, 2021-11-06:


impeeza, 2021-11-06:
¿?

urherenow, 2021-11-06:
Dude made it abundantly clear that he can't follow directions, starting with deleting the template instead of using it, I'll bet even reading it.

OP: you can't simply inject the new .bin. You have to extract ALL of the files too, overwriting the old ones.

impeeza, 2021-11-06:
Ah, I now understand, that error is for not read the release and follow the instructions for updating.

Not a bug or an issue!

This thread must be closed

Marshmellin, 2021-11-07:
I don’t understand English very good so this is confusing for my. I’m gonna try it later but I literally renewed the bin 1000 times.

urherenow, 2021-11-07:
That's your problem. You don't "renew the bin". You extract the entire .zip file, overwriting all other files.

There is 1 and only 1 thing that causes this error. The hekate.bin your injecting is newer than the nyx.bin on your card.

Marshmellin, 2021-11-07:
Ok so I need to inject a newer version of the nyx
Or I need to inject a older version of hekate?

urherenow, 2021-11-07:
You need to go to the releases page, download the .zip file, and extract everything inside, onto the root of your sd card.

Marshmellin, 2021-11-07:
Thank you so much. Right now I can't check it but when I can I'll write to you

urherenow, 2021-11-07:
Please don't here. Go to discord or gbatemp, and close this issue. This space is for reporting bugs, it's not a support forum.

Marshmellin, 2021-11-07:
What discord?

Marshmellin, 2021-11-07:
Can you give me the link?

impeeza, 2021-11-07:
Literally any web browser will give you a link as answer.

urherenow, 2021-11-07:
ReSwitched or RetroGamer

impeeza, 2021-11-07:
Gbatemp

CTCaer, 2021-11-07:
Dunno why that convo reached that point, but just extract the bootloader folder from release zip into sd card.
No matter how many times you replace the binary, you will always get the same message if you don't update that folder.

closed 2021-11-07 by CTCaer