CTCaer/hekate #680: Strange behaviours on OLED

shadow2560, 2021-11-09:
I've made some tests on OLED and it seems to have some problems for reading or writing to the nand (block read/write errors or emmc not detected, mounting on a PC via USB have the same problems).

Depending on the length of the wires at points B and C the behavior may change. With short wires I have block errors but with long wires the nand is not detected but, in this case, if I launch Atmosphere on sysnand with it payload it works without problems, it backup the CAL0 and biskeys and launch homebrews, games, etc... normaly (only with it payload, if I try to launch it via a Hekate config Hekate says that the emmc is not detected).

I've tested with SD in FAT32 and EXFAT, with or without Minerva, with or without Nyx but I can't find how to fix this, these randoms behaviours are realy strange and I can't find if it's a problem in Hekate like the test with long wires seems to confirm or a problem from somewhere else like the short wires test seems to show.

CTCaer, 2021-11-09:
That's basically *Signal Integrity 101*

tl;dr: your setup fails to work on HS400 (aka 200MHz DDR).

Mind you that yes I can add support for what HOS supports if it fails that,
(which is try HS 400 -> then try HS52 -> then try 1bit HS52)
but that will only mitigate your actual issue.

And that issue, as I said, is that your eMMC can't work on 400 MB/s bus speed but it will fall back to 52 MB/s and worst case scenario at 6.5 MB/s.

Additionally, if I implement that, it will still be taken like with broken sd card connectors: Nyx will inform you about that HW issue with a pop up.

To sum up, ok I will add that support. BUT it will be considered a HW fault. And such cases should be repaired.

shadow2560, 2021-11-09:
OK, thank. I'm not sure that I have understand evrything cause it's very technical. For you what should I do to fix that hardware problem?

Masamune3210, 2021-11-10:
There is too much interference or other issues which is causing the NAND to fail reads/writes. Either your wires are bad quality, you have them routed near noisy components or they are too long/wrong thickness. Something is causing interference in any case, the solution is to find the cause of the interference and resolve it

Leerz, 2022-01-17:
> I've made some tests on OLED and it seems to have some problems for reading or writing to the nand (block read/write errors or emmc not detected, mounting on a PC via USB have the same problems). [...]

having the same issues, emmc benchmark fails, may points are solid on clk and dat0 which are the problematic ones.
I've used both sxcore and hwfly and the same, even routed rst on the bottom instead of all the way to the fan side.

I've managed to make 1 good dump afterwards, it can't dump anymore getting read block errors.

I'm thinking which line is causing it

I'm using enameled jumper wires. 03

CTCaer, 2022-01-20:
Latest release adds support for lower clocks.

**REMINDER: that is considered still a HW issue, even if you know can boot!**

closed 2022-01-20 by CTCaer