CTCaer/hekate #681: Question about booting OFW over Hekate (fuses burning/kernel panic)

ErbayI, 2021-11-13:
It is a bad idea to boot OFW over Hekate because fuse Burning? 

As I understood:
When booting normal than it will burn fuses on a update but over Hekate (RCM) it will skip fuse check/burning, so no fuse will burnt.

So when i start OFW over Hekate und update the firmware, will i get a kernel panic error when i boot normal because no fuse was burnt on the previous update? 

If it is like that:
Gives a setting in Hekate or in the "hekate_ipl.ini" to configure that when booting in OFW, that it will start it as regular like normal Switch?

My Switch is the v1 model with a trinket m0 installed and I want my sysNAND clean for online, for homebrew i use the emuNAND 
 
[image]

urherenow, 2021-11-13:
Github is not a support forum. 

The second you boot OFW *without using hekate/Atmosphere*, your fuses will be burnt. Doesn't matter how the upgrade was installed.

impeeza, 2021-11-13:
A) this is a isues/bug reporing space.  For help please go to proper spaces: discord gba temp or use your favorite search engine.

B) read above.

C) clase this issue soon please

closed 2021-11-13 by ErbayI