CTCaer/hekate #682: failed to launch hos because exfat

hahnmt, 2021-11-21:
Running ofw 13.1.0. Cfw 10.2.0 with exfat drivers. Exfat few ver Confirmed using nxnandmanager. 

Setup using emummc migrated from sxos. 

Receive error on cfw launch stating that disk is exfat but hos is using fat32 driver. 

Saw this prior closed issue #480 which sounds very similar 

Ofw boots just fine.

Masamune3210, 2021-11-21:
If HOS is updated without installing the exfat driver then all you should have to do is install the exfat version of the firmware, but you really should use fat32 unless you absolutely have to use exfat for some reason. HOS's exfat driver is pure garbage and causes all sorts of issues all the time making it unstable at best and damaging at worst

hahnmt, 2021-11-22:
Thanks for the reply. Just to be clear the 10.2.0 was working on sxos running exfat formatted prior to conversion. Since upgrading ofw to 13.1 I have switched to atmosphere and was surprised to find the HOS launch error. 

To confirm, I browsed the SYSTEM partition of eMMC and confirmed the presence of the exfat driver nca in the committed folder. 

I have not yet tested the eMMC on a FAT32 SD to see how it behaves

Masamune3210, 2021-11-22:
Try booting with fusee directly to see if it will boot, something is confusing hekate into thinking that the exfat driver is not installed, and triggering that safety. Its there to keep people from trying to boot without the driver installed and failing when boot is passed to HOS

hahnmt, 2021-11-26:
I have confirmed eMMC BCPKG2-1-Normal-Main contains the wrong title (fat32 instead of exFat).  while the file system contains the exFat setup and drivers. 

Fusee halts me with a similar error. I assume it is using the same title check logic to check package2 title is 010000000000081b

hahnmt, 2021-11-26:
Ok I'm closing this.  going to say the issue has more to do with the history of the firmware coming from sxos than atmosphere / hekate

hahnmt, 2021-11-26:
Not related to hekate

closed 2021-11-26 by hahnmt