CTCaer/hekate #685: Windows will detect UMS device but can't assign a letter

EpicLPer, 2021-11-30:
Heya!

I've browsed the issues a bit but couldn't find this particular weird case here.  
Plugged the Switch in, started UMS and while the device and driver is correctly recognized and being displayed it doesn't actually show any partitions despite saying the drive is Online.  

[image]  

[image]

Could this be due to an incompatible SD Card? Tho would be weird since everything else seems to work perfectly normal (EmuMMC, Android, Backups, etc.).

Thanks for help already!

CTCaer, 2021-12-02:
You should fix your windows installation first.
That error in disk management is not ok.

After that, try again and if you still have the problem, *reopen the issue*.

(Have in mind that probably I won't be able to do anything. That's because it seems like a host issue. But we'll see after you fix that.)

closed 2021-12-02 by CTCaer

EpicLPer, 2021-12-05:
I hope commenting doesn't automatically open the issue again but yes, you were right and sorry for opening one too early :(

Just to give details on how/why this happened: I had VMware running in the background to mess with the (old) Android partition files, and apparently even if you deny the automatic USB connection prompt you get to attach the USB device to the VM it'll still not fully be recognized by host. I had to fully close VMware to fix this.