CTCaer/hekate #686: Partition Manager backup/restore breaks installed games (unsets archive bits)

Chariblaze, 2021-12-01:
Running Hekate 5.6.5. Testing with SysMMC. Example:
1. Some small games are installed to the SD card (test: Twitch app, 31 MB).
2. Partition Manager does the auto backup/restore for <1 GB SD contents. Tested with two layouts: all HOS, and 4 GB emuMMC.
3. The archive bits on the *.nca folders become unset.
4. The installed games error out when trying to launch.
5. Using the Fix Archive Bit tool resolves the issue.

---

Archive bits unset after partitioning/formatting:

```
F:\>attrib Nintendo\Contents\registered\* /S /D
A                    F:\Nintendo\Contents\registered\00000049\b5bbbbc0f879f56142001a4274f5530f.nca\00
                     F:\Nintendo\Contents\registered\00000049\b5bbbbc0f879f56142001a4274f5530f.nca
A                    F:\Nintendo\Contents\registered\00000070\3239d4ec3a05c301dd747a796d1eaabc.nca\00
                     F:\Nintendo\Contents\registered\00000070\3239d4ec3a05c301dd747a796d1eaabc.nca
A                    F:\Nintendo\Contents\registered\000000A5\9dab96a5bd219a4d483c52d0d64b3402.nca\00
                     F:\Nintendo\Contents\registered\000000A5\9dab96a5bd219a4d483c52d0d64b3402.nca
A                    F:\Nintendo\Contents\registered\00000022\cfef2dee0680fe8565c05b2f8f97d233.nca\00
                     F:\Nintendo\Contents\registered\00000022\cfef2dee0680fe8565c05b2f8f97d233.nca
A                    F:\Nintendo\Contents\registered\000000DD\1e1b8a235e65710cdbd3a2ea6e81cc5c.nca\00
                     F:\Nintendo\Contents\registered\000000DD\1e1b8a235e65710cdbd3a2ea6e81cc5c.nca
                     F:\Nintendo\Contents\registered\00000028
                     F:\Nintendo\Contents\registered\00000048
                     F:\Nintendo\Contents\registered\00000049
                     F:\Nintendo\Contents\registered\0000004A
                     F:\Nintendo\Contents\registered\00000052
                     F:\Nintendo\Contents\registered\00000053
                     F:\Nintendo\Contents\registered\00000065
                     F:\Nintendo\Contents\registered\00000068
                     F:\Nintendo\Contents\registered\00000069
                     F:\Nintendo\Contents\registered\00000070
                     F:\Nintendo\Contents\registered\00000075
                     F:\Nintendo\Contents\registered\00000082
                     F:\Nintendo\Contents\registered\00000087
                     F:\Nintendo\Contents\registered\0000008F
                     F:\Nintendo\Contents\registered\00000090
                     F:\Nintendo\Contents\registered\0000009A
                     F:\Nintendo\Contents\registered\000000B6
                     F:\Nintendo\Contents\registered\000000CC
                     F:\Nintendo\Contents\registered\000000DF
                     F:\Nintendo\Contents\registered\000000F2
                     F:\Nintendo\Contents\registered\000000A5
                     F:\Nintendo\Contents\registered\00000022
                     F:\Nintendo\Contents\registered\000000DD
```

[image]
					 
[image]

&nbsp;
Archive bits after running fix tool:

```
F:\>attrib Nintendo\Contents\registered\* /S /D
A                    F:\Nintendo\Contents\registered\00000049\b5bbbbc0f879f56142001a4274f5530f.nca\00
A                    F:\Nintendo\Contents\registered\00000049\b5bbbbc0f879f56142001a4274f5530f.nca
A                    F:\Nintendo\Contents\registered\00000070\3239d4ec3a05c301dd747a796d1eaabc.nca\00
A                    F:\Nintendo\Contents\registered\00000070\3239d4ec3a05c301dd747a796d1eaabc.nca
A                    F:\Nintendo\Contents\registered\000000A5\9dab96a5bd219a4d483c52d0d64b3402.nca\00
A                    F:\Nintendo\Contents\registered\000000A5\9dab96a5bd219a4d483c52d0d64b3402.nca
A                    F:\Nintendo\Contents\registered\00000022\cfef2dee0680fe8565c05b2f8f97d233.nca\00
A                    F:\Nintendo\Contents\registered\00000022\cfef2dee0680fe8565c05b2f8f97d233.nca
A                    F:\Nintendo\Contents\registered\000000DD\1e1b8a235e65710cdbd3a2ea6e81cc5c.nca\00
A                    F:\Nintendo\Contents\registered\000000DD\1e1b8a235e65710cdbd3a2ea6e81cc5c.nca
                     F:\Nintendo\Contents\registered\00000028
                     F:\Nintendo\Contents\registered\00000048
                     F:\Nintendo\Contents\registered\00000049
                     F:\Nintendo\Contents\registered\0000004A
                     F:\Nintendo\Contents\registered\00000052
                     F:\Nintendo\Contents\registered\00000053
                     F:\Nintendo\Contents\registered\00000065
                     F:\Nintendo\Contents\registered\00000068
                     F:\Nintendo\Contents\registered\00000069
                     F:\Nintendo\Contents\registered\00000070
                     F:\Nintendo\Contents\registered\00000075
                     F:\Nintendo\Contents\registered\00000082
                     F:\Nintendo\Contents\registered\00000087
                     F:\Nintendo\Contents\registered\0000008F
                     F:\Nintendo\Contents\registered\00000090
                     F:\Nintendo\Contents\registered\0000009A
                     F:\Nintendo\Contents\registered\000000B6
                     F:\Nintendo\Contents\registered\000000CC
                     F:\Nintendo\Contents\registered\000000DF
                     F:\Nintendo\Contents\registered\000000F2
                     F:\Nintendo\Contents\registered\000000A5
                     F:\Nintendo\Contents\registered\00000022
                     F:\Nintendo\Contents\registered\000000DD
```

CTCaer, 2021-12-02:
That's actually weird, because I made sure to copy over the attributes:
https://github.com/CTCaer/hekate/blob/master/nyx/nyx_gui/frontend/gui_tools_partition_manager.c#L179
This basically sets the copied file's attributes to the attributes of the source file. And 0xFF mask allows every attribute from the original to be set on the new file.
```c
FRESULT f_chmod (
  const TCHAR* path, /* [IN] Object name */
  BYTE attr,         /* [IN] Attribute flags */
  BYTE mask          /* [IN] Attribute masks */
);
```
```
f_close(&fp_dst);                 // Syncs and closes destination file
f_chdrive(dst);                   // Changes main disk to destination, so relative path based functions like f_chmod work
f_chmod(path, fno.fattrib, 0xFF); // sets attributes to the dst file from src file and uses relative path.
```

Hmm...

Chariblaze, 2021-12-02:
> Btw just making sure, I assume you had correct archive bits before the backup/restore, right?

Yeah. I've also had Nyx set/unset the bits on its own in a fun loop:

1. Partition/format with backup & restore
2. Fix archive bits (5 set)
3. Partition/format with backup & restore
4. Fix archive bits (5 set)
5. ...

Chariblaze, 2021-12-02:
It's specifically folders that don't have their attributes copied, files are fine.

1. `F:\>attrib +A /s /d *`
2. `F:\>attrib /s /d * > attribs-before-format.txt`
3. Unmount, partition/format in Nyx.
4. `F:\>attrib /s /d * > attribs-before-format.txt`
5. Compare txts with Notepad++ Compare plugin or something similar.

CTCaer, 2021-12-03:
Found the issue in the code and it's exactly as described.
Thanks for reporting that and actually investigating it.

Will be fixed in next version.

CTCaer, 2022-01-20:
Get latest release that fixes that.
Thanks for testing and actually finding the issue on your own

closed 2022-01-20 by CTCaer