CTCaer/hekate #687: Touchscreen not working. Stuck at Enter Date and Time for Nyx

uberyoji, 2021-12-02:
I am currently using v5.6.5 and Nyx v1.1.1 on a Switch lite with a hwfly modchip. When I boot into hekate, the touchscreen is disabled but Nyx is not frozen since the temp and time will update in the background.

Is there a way to disable the Enter Date... step?

Also no buttons are working.

Fortunately I can auto boot into Atmosphere but I cannot use any of the great features of Hekate :-/ such as nand backup.

I am available to test.

uberyoji, 2021-12-02:
Here is my hekate_ipl.ini:

[config]
autoboot=1
autoboot_list=0
bootwait=3
backlight=100
autohosoff=0
autonogc=1
updater2p=0
bootprotect=0

[Atmosphere CFW]
payload=bootloader/payloads/fusee.bin
icon=bootloader/res/icon_payload.bmp
kip1patches=nosigchk

[Lockpick RCM]
payload=bootloader/payloads/Lockpick_RCM.bin
icon=bootloader/res/icon_payload.bmp

[Warmboot Error Fix]
fss0=atmosphere/package3
stock=1
emummc_force_disable=1
icon=bootloader/res/icon_switch.bmp

CTCaer, 2021-12-02:
Does your touchscreen work inside HOS?
(Btw, you can only use touch or joycon in Nyx. And the Lite's joycon are not supported because I don't have a Lite to reverse their function and make a driver.)

(Also Warmboot Error Fix? That's Stock aka OFW btw. But anyway.)

uberyoji, 2021-12-02:
Arg. I think my touch screen is broken. Doesn't seem to work when i try it in the setting's input test.... Now i need to open it up and see if the digitizer cable is unseated...

Also I will remove the warmboot section.

I think you can close this. Thanks for the quick response.

masagrator, 2021-12-02:
@uberyoji why not close it yourself?

uberyoji, 2021-12-02:
Oh sure. I assume I was not allowed.

closed 2021-12-02 by uberyoji