CTCaer/hekate #688: About translation

yyoossk, 2021-12-06:
I want to translate it into Japanese
Doesn't it support language files?

yyoossk, 2021-12-06:
I tried the translation, but the characters were garbled
Can't change the font file?

CTCaer, 2022-07-07:
https://github.com/CTCaer/hekate/issues/252
mentioning main thread and closing duplicate

closed 2022-07-07 by CTCaer

TheTokenMoose, 2024-01-26:
> The gui based on [lvgl](https://lvgl.io/), build your own font c file(see this [interui_30.c](https://github.com/CTCaer/hekate/blob/master/bdk/libs/lvgl/lv_fonts/interui_30.c) as template) and bitmap(res.pak in release zip file, start with 0x7900) with Japanese words, enable [LV_TXT_UTF8](https://github.com/CTCaer/hekate/blob/master/bdk/libs/lv_conf.h#L129), add [custom font](https://github.com/CTCaer/hekate/blob/master/bdk/libs/lv_conf.h#L206) and recompile.
> 
> This is my work. I use LvglFontTool and HarmonyOS Sans font to build a c file font. And replace the default font from interui_30 to this.
> 
> [image] [image]

My Hekate UI is in Chinese, like this. I'm wanting to change it to English without a fresh install. Is it possible to reverse this type of change by just replacing the files on my SD card with the same version source files? I hope this isn't a dumb question, I'm new to this..

easyworld, 2024-01-27:
> My Hekate UI is in Chinese, like this. I'm wanting to change it to English without a fresh install. Is it possible to reverse this type of change by just replacing the files on my SD card with the [...]

Simply replace `/bootloader/sys/res.pak` with the version from the corresponding official zip file.

TheTokenMoose, 2024-01-28:
> > > The gui based on [lvgl](https://lvgl.io/), build your own font c file(see this [interui_30.c](https://github.com/CTCaer/hekate/blob/master/bdk/libs/lvgl/lv_fonts/interui_30.c) as template) and bitmap(res.pak in release zip file, start with 0x7900) with Japanese words, enable [LV_TXT_UTF8](https://github.com/CTCaer/hekate/blob/master/bdk/libs/lv_conf.h#L129), add [custom font](https://github.com/CTCaer/hekate/blob/master/bdk/libs/lv_conf.h#L206) and recompile.
> > > This is my work. I use LvglFontTool and HarmonyOS Sans font to build a c file font. And replace the default font from interui_30 to this.
> > > [image] [image]
> > 
> > 
> > My Hekate UI is in Chinese, like this. I'm wanting to change it to English without a fresh install. Is it possible to reverse this type of change by just replacing the files on my SD card with the same version source files? I hope this isn't a dumb question, I'm new to this..
> 
> Simply replace `/bootloader/sys/nyx.bin` with the version from the corresponding official zip file.

Thank you, that worked instantly. Appreciate it.

xElChoco, 2024-04-17:
Hi TheTokenMoose, i'm also trying to change my chinese hatake to english. Can you please explain me a little more how did you managed to do it?

TheTokenMoose, 2024-04-17:
> Hi TheTokenMoose, i'm also trying to change my chinese hatake to english. Can you please explain me a little more how did you managed to do it?

I downloaded the official zip file, opened it and copied the file '''/bootloader/sys/nyx.bin'' onto my SD card replacing the same file. This solved it for me immediately.