CTCaer/hekate #689: Creating Android partition causes SD UMS incompatibilty with Android phone

zemixe, 2021-12-09:
I noticed recently that SD UMS was no longer working when connected to my Android 11 device so I tested creating HOS, emuMMC, Linux and Android partitions and I found that the SD can mount with all of them except when an Android partition is created. If an Android partition is present then the device simply wants to format the card: 

[image]

Is it possible to fix that in hekate or is it unavoidable? Thank you.

CTCaer, 2021-12-12:
That's unavoidable. Android on Switch needs Hybrid MBR - GPT combo.
That's to allow Android and HOS booting.

Ask your phone's vendor to have proper support for Hybrid MBR and not do whatever it does now.
Which is wrong btw, because hybrid mbr if not supported, it should be at least parsed as GPT as per spec.
It basically actively identifies hybrid mbr and rejects the whole storage.

closed 2021-12-12 by CTCaer

zemixe, 2021-12-12:
Ah thank you for explaining. Thanks Samsung!

Masamune3210, 2021-12-13:
Why does it being samsung not surprise me....