CTCaer/hekate #690: Mariko android/Lakka/Linux

lmm1191, 2021-12-12:
I know this was already asked but the thread is closed so most likely no one will provide answers there anymore. I've read that sx core firmware has already been dumped, maybe a work around can be found on the script to avoid getting the device bricked in using coreboot.rom?

zemixe, 2021-12-12:
"a work around can be found on the script" What are you talking about here? 

1. there's already protection in place so that you can't just brick a mariko unit by attempting to use coreboot
2. a lot of work is required to support android, lakka and linux on mariko, you don't just flip a switch and it has basically nothing to do with sx core firmware
3. this repository is for reporting hekate issues/requesting hekate features etc

lmm1191, 2021-12-12:
> "a work around can be found on the script" What are you talking about here?
>
> 1. there's already protection in place so that you can't just brick a mariko unit by attempting to use coreboot
> 2. a lot [...]

Requesting hekate features, exactly

masagrator, 2021-12-12:
This is not a request for Hekate feature. Implementing support for Linux and Android is on the side of Android and Linux maintainers, not Hekate's.

CTCaer, 2021-12-12:
This was already answered many times before.
If support comes, you will learn it.

And yeah, it's not about hekate supporting mariko (it already does) but the whole android/ubuntu boot chain.

closed 2021-12-12 by CTCaer