CTCaer/hekate #692: help error emummc

putra1990, 2021-12-28:
emummc is forced but not enabled failed to launch hos

urherenow, 2021-12-28:
Discord or the gbatemp forums are far better places to seek help. Github is for collaborating and submitting bug reports, whereas your issue is 99% likely to be either your fault, or a hardware problem (sd card socket pins, fake sd card, etc...). Nobody can really guess, since you've provided basically no information at all. Make a post at gbatemp, detailing your info (fw version, hekate+Atmosphere versions, Switch model, SD card brand, capacity and what file system it's formatted to, the directory structure on your card, the contents of hekate_ipl.ini, the contents of /emuMMC/, etc...

CTCaer, 2022-01-04:
you either never created an emummc or you have it disabled as the error says.

closed 2022-01-04 by CTCaer