CTCaer/hekate #694: Hekate dont want to cooperate with different SD cards except one 64 GB

airboozik, 2022-01-02:
Hello,

I am looking for a solution as a fresh in the topic.

I made everything according to the tutorials to boot to the hekate and to the atmospere. I managed tinfoil etc..
Mod hwfly

Problem was thst my card was too small 64 GB samsung to keep all the data that i needed. 

I decided to swap to 256 kingstone canvas..

I did exact the same procedure to launch bootlader but i cannot reach to hekate.. I also tried third 2gb card and the same problem..

Hwfly blinks for a long time but the one thing i See is black screen.. 

Do you know what could be the issue? 
I am thinking that something will be bad with filesystem or something but i dont know what to check.. 

And still when i go back to 64GB it starts In a second..

Masamune3210, 2022-01-02:
Make sure the card is MBR partition style, that trips up a bunch of people when it comes to console modding

airboozik, 2022-01-02:
I checked it Today for Maybe 50 Times. I have new info that one of my SD cards, 4 GB probably sdhc works also flawless..

closed 2022-01-04 by CTCaer

Masamune3210, 2022-01-04:
My bad. I totally missed the fact that it was before rcm was even entered