CTCaer/hekate #695: 1TB micro sdxc card partition manager issue

Paddy31, 2022-01-04:
Hi CTCaer,
before everything, I wish you pleasant and happy new 2022 year!
Many thanks for the huge amount of work around hakate and others.

I use 1 TB PNY micro sdxc card (see attached picture) inside my switch with 11.0.0 firmware. When entering partition manager and after formatting or creating emuMMC partion, I get error messages during emuMMC file creation or launching atmosphere 1.2.5 through emuMMC.

Analysis of the micro sdxc car under linux mint 20.2 gnome-disks shows a first empty partition of 17MB then 1TB, the 31GB emuMMC partition and finally another empty partition of 2,1MB (when emuMMC partition created).
See attached picture where a first empty partition of 17MB and 1TB exFAT is shown (my last try).

I don't know if it could be a hekate 5.6.5 issue or may be something to do better on my side? 
I add, I never met this issue using 64GB or 128GB sdxc cards before (those cards are SAMSUNG EVO or anonymous marks) but less capacity than 1TB.

Many thanks for your help and advices CTCaer (^_^)'
Let me know if I can send you element to also help?

[image]
[image]
[image]
[image]
[image]
[image]

Conan179, 2022-01-04:
1tb? "PNY"? How much did you spend on the card?
smells strongly of fake card!

Paddy31, 2022-01-04:
I leave you the seller link : https://fr.aliexpress.com/item/1005003719286719.html
This 1TB card works very well both on android and linux mint.

Please, see other 3 sdxc cars where hakate works fine. 

[image]

Conan179, 2022-01-04:
20€ for 1tb...
Dude you have a fake card, you muss use a legit card! Hekate is not the problem!

Paddy31, 2022-01-04:
How to verify fake card please?

Conan179, 2022-01-04:
https://www.heise.de/download/product/h2testw-50539

Paddy31, 2022-01-04:
Thanks a lot.

Here is the verification result: it seems to be OK

Warning: Only 1024 of 1023749 MByte tested.
Test finished without errors.
Reading speed: 1.04 GByte/s
H2testw v1.4

[image]

Conan179, 2022-01-04:
only 1gb from 1024gb are testing....
You see the programm cand test the card, she is ne fake card.

masagrator, 2022-01-04:
Testing only 1024 MB is nowhere near enough. You must clean whole sdcard.

To do that you must format sdcard before testing.

urherenow, 2022-01-04:
To be fair, hekate's partition manager would have formatted the card. If not, I'm pretty sure there is a format option in there.

Also, there is a section for showing the sod info. Get a picture of that info and post it here. There are several signs that make it blatantly obvious that it's fake. Of course, at that price, we are already certain that it is fake already.

closed 2022-01-04 by CTCaer

Paddy31, 2022-01-08:
Many thanks to all for your answers and precious time to enhance my comprehension.

@CTCaer: may be hekate can/will be able to advice users the micro sd card is fake?
I began to format PNY fake sd car under linux mint and after hekate partition creation, the partition map (using GParted) shows a starting 64Mo partition. This can be a way to quickly check directly inside hekate the sd card is fake (or doesn't give satisfaction)

**Last, but not least, BIG THANKS to you CTCaer (^_^)**

[image]
?

urherenow, 2022-01-08:
Thi issue was closed ang he already told you where to check.

Console Info -> SD card

Within NYX

CTCaer, 2022-01-08:
> Thi issue was closed ang he already told you where to check.
> 
> Console Info -> SD card

This ^

It's the only way to know if sd is fake and it's not as simple as doing status heuristics.
It's extremely easy to detect by human eye though.

As for that 64MB in front it has nothing to do with that. That's a performance trick to align the first partition to AU size.
Data in there can't be used to identify if a sd is fake.
Only corrupted/missing data for no reason and sd info can be used for that

Paddy31, 2022-01-08:
I know the issue is close. My message was to thank everyone learn and answer me around sd card formatting and I try also to humbly participate. Sorry for being not enough smart, I apologize. 

After Linux Mint exFAT formatting,  Console Info returns available sd card. I didn't make picture but 999Gb is shown with a lot of available space. At first sight, all seemed good. That's why I opened this closed ticket...

H2testw (after very long hours checking) reports errors while checking is not achieved at this moment.
Warning: Only 1023965 of 1024000 MByte tested.
**The media is likely to be defective.**
14.5 GByte OK (30601155 sectors)
121.0 GByte DATA LOST (253890621 sectors)
Details:0 KByte overwritten (0 sectors)
0 KByte slightly changed (< 8 bit/sector, 0 sectors)
**121.0 GByte corrupted (253890621 sectors)**
0 KByte aliased memory (0 sectors)
First error at offset: 0x000000012f2c7010
Expected: 0x8735211b760e7110
Found: 0x8735211b760e6110
H2testw version 1.3

Many thanks again to enhance my knowledge, have a nice week-end.